bigtreetech/BIGTREETECH-SKR-mini-E3 #78: Firmware crashes after attempting to connect to octoprint

Kamehamehaas, 2019-10-31:
I've had my skr mini e3 for about a week and wanted to enable make some adjustments to make the prints better. Long story short I changed some Auto bed leveling settings in the firmware, flashed it to the board, and now anytime I go to connect to octoprint my printer becomes unresponsive and restarting it causes the screen to go blank and still not be unresponsive. Reflashing the board fixes it until I try to connect to octoprint.

Not sure what I might have done or accidently done to cause this and looking for what to look for. Also I do have tape covering the power pin on the USB cable so that isn't the cause/solution.

MarcusDLG, 2019-10-31:
How big is your firmware.bin file size? I had an issue where my firmware would flash, work for a little bit, but with every button push I got closer to a magical jack in the box moment where the printer would freeze. Exact same issue as you, I'm running a bltouch 3.1. I started there because my problems originated after I installed the bltouch.

It wasn't the issue.

Long story short, your firmware should be 224kb. If it's not, mine was 229kb, check #39 and change the eeprom start address.

Kamehamehaas, 2019-11-01:
My file size was also 229kb....

So I tried the fix on the link you shared. Specifically the following:
`change line 38 that reads
#define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 1024 - 2 * EEPROM_PAGE_SIZE)

to:
//#define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 1024 - 2 * EEPROM_PAGE_SIZE) //use for F103RC
#define EEPROM_START_ADDRESS uint32(0x8000000 + 256 * 2048 - 2 * EEPROM_PAGE_SIZE) //use for F103RE`

However two things I noticed. 
My default line looks like this:
`#define EEPROM_START_ADDRESS uint32(0x8000000 + STM32_FLASH_SIZE - 2 * EEPROM_PAGE_SIZE)`

Which might be why trying the fix above didn't work? I tried both addresses. the first one caused the firmware to never load in the first place. The second loaded but still crashed.

However your comment made me resize the code (enabled slim LCD) and it is working now.

I would still love to have everything work as designed without the slim LCD so any help to get me there would be appreciated