bigtreetech/BIGTREETECH-SKR-mini-E3 #79: BLTouch bed leveling not working well.

pandabear41, 2019-10-31:
I have configured the SKR mini E3 v1.2 to use the BL touch. The FW works fine and probe seem to function correctly. I setup my Z offset correctly. When I go to print however the nozzle is too far from the build plate on one side and almost crashes on the other side. Somewhere in the middle it seems to be fine. It's like the bed leveling isn't working at all, but that isn't the case. I see the z screw moving ever so slightly, but it's like something isn't working correctly. I've tried adjusting probing speeds and probe distances from edges and nothing seems to work very well. 

Is anyone else having this problem? Any tips to fix this issue?

I will post pics when I get home.

[Config.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3794397/Config.zip)

brew99, 2019-10-31:
try removing the defining of the z-probe offset here: (i.e. make it 0)
#define NOZZLE_TO_PROBE_OFFSET { 48, -2, -2 }

and add the Z-probe offset to your start G-code in the slicer using M851 Z-2

Ensure you have the G29 after the G28 in your start g-code

This is how I use it with my BLtouch and have no issues (I also use the "probe" pins and change to PC14 in the pins file)

pandabear41, 2019-11-06:
Fixed it by enabling RESTORE_LEVELING_AFTER_G28. 
I was saving mesh before a batch print job and it wasn't enabling automatically after homing all axis.

closed 2019-11-06 by pandabear41