bigtreetech/BIGTREETECH-SKR-mini-E3 #81: HotEnd fan always on

pansk, 2019-11-01:
Tried with:
- default firmware
- firmware downloaded from here
- latest marlin 2.0.x with patches applied according to documentation with FAN_SOFT_PWM option
- latest marlin 2.0.x with patches applied according to documentation without FAN_SOFT_PWM option

My hotend fan (on a 1.2 board) remains always turned on, despite setting it to any value from 0% to 100%.

bojanpotocnik, 2019-11-01:
Assuming you wired it correctly, this fan is always on by default - it is connected directly to the power supply (24 V), the same as the housing fan. You control PWM only of the FAN0 fan.
[image]

damvcoool, 2019-11-02:
I followed Teaching Tech video and i do no have this issue...maybe you should try it?
https://www.youtube.com/watch?v=-XUQKQnUNig

pansk, 2019-11-02:
Sorry for the confusion, I obviously meant the cooling fan.
To be perfectly clear: all the fans of my printer are always on, no matter what fan speed I specify.

Wiring it incorrectly (from a stock board) is virtually impossible, as the connectors have the same shape and are in the same position... I detached them one by one from the stock board and placed  in the appropriate position on the skr mini e3.

bojanpotocnik, 2019-11-02:
> Wiring it incorrectly (from a stock board) is virtually impossible, **as the connectors have the same shape and are in the same position...** I detached them one by one from the stock board and placed in the appropriate position on the skr mini e3.

Just to double-check, if you connected the fans exactly as on the stock board (as you wrote in the bold sentence) then you wired it incorrectly, because the FAN1 and FAN0 connectors placement is exchanged with each other compared to the stock board ([Fan0 and Fan1 mixed compared to stock Ender 3](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/16)).

mfoco, 2019-11-06:
Ok, I will double-check (next week), thank you!

pansk, 2019-11-06:
Ok, I'll double check, thank you!

nachosan100, 2019-11-10:
i found it to be working in a very small tolerance band: 1% starts the hotend fan, 3% already 100% powers it - so it must be a scaling issue from marlin itself....anybody have an idea what setting is causing this?

nachosan100, 2019-11-10:
it can be fixed by playing with
#define FAN_MIN_PWM 1 (original 50)
#define FAN_MAX_PWM 5 (original 128)
figures in configuration_adv.h

pansk, 2019-11-11:
I was away for the whole last week. I'll try both solutions tonight, and if anything solves I'll post the solution here.

pansk, 2019-11-25:
> Just to double-check, if you connected the fans exactly as on the stock board (as you wrote in the bold sentence) then you wired it incorrectly, because the FAN1 and FAN0 connectors placement is exchanged with each other compared to the stock board (Fan0 and Fan1 mixed compared to stock Ender 3).

And yes, you were right, @bojanpotocnik. Closing this.

closed 2019-11-25 by pansk

emiliorizzo, 2020-09-27:
The motherboard has only 2 CNC fans.
The standard connection is: Part Fan -> Fan0, Motherboard Fan -> Fan1. The hotend fan is wired directly without CNC.

I have an Ender-3Pro wired in this way. I can control the part's fan from the LCD screen. The motherboard fan turns on only when the steppers are running.

I think if you want to use the temperature controlled hotend fan, the only way is to use FAN1, connect the motherboard fan directly, and modify the firmware to use this setup.