bigtreetech/BIGTREETECH-SKR-mini-E3 #84: Cant compile firmware for mini E3 1.0 DIP with 2208 drivers

gannon145, 2019-11-05:
When changing settings to 2208 drivers I can't complie marlin 2.0 firmware and get the following:

"TMC2208 or TMC2209 on X requires X_HARDWARE_SERIAL or X_SERIAL_(RX|TX)_PIN."

[image]

gannon145, 2019-11-05:
Readme says to use:

#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BIGTREE_SKR_MINI_E3
#endif

Can you explain why this is the wrong repo, sorry I don't understand this one.

diabl0w, 2019-11-05:
if you have the DIP board, then I am not sure why you are in the E3 mini repo and trying to compile for the E3 mini board... https://github.com/bigtreetech/BIGTREETECH-SKR-E3-DIP-V1.0

`#ifndef MOTHERBOARD
  #define MOTHERBOARD BOARD_BIGTREE_SKR_E3_DIP
#endif`

gannon145, 2019-11-05:
Ahh got it, there is my point of confustion. It was labeled as SKR mini E3 DIP, so I assumed the mini was mor important than the DIP.

Thank you!

closed 2019-11-05 by gannon145