bigtreetech/BIGTREETECH-SKR-mini-E3 #86: Linear Advance

PLAyerNos, 2019-11-06:
Just tried Linear Advance with Classic Jerk, X and Y steppers jump and leave artefacts on the walls for 10-15 first layers. Jerk can't be controlled, no matter what I set feels very aggressive.
Corners are much better but I wouldn't use it for the reasons above.

Used with Junction Deviation prints much better than without , can see difference on the corners when changing K value.
Extruder stepper  going crazy with back and forward movements making rattling noise.
I don't think it should do that. 
Do you experience same problems?
Hope not!!!

bojanpotocnik, 2019-11-06:
> Extruder stepper going crazy with back and forward movements making rattling noise.

This also happens to me, however I think it is normal. The motor does a slight retract - pressure release (which depends on the K factor) for every change of the linear movement direction. It is true that the printer is not as silent anymore (especially when printing detailed parts without any long linear movements - this really becomes a constant rattling noise), but the prints are better and motor is heating up no more than usual (without linear movement).
I received the direct drive extruder and hotend today and will mount it over the weekend - this effect shall be minimal there (the K factor is much smaller).

PLAyerNos, 2019-11-06:
I hear you but if you try LA with classic jerk, extruder is very silent, as normal.
It doesn't go back nearly as much as 4mm retraction. Just push less material I guess.

PLAyerNos, 2019-11-12:
No Lin Advance for me then...
Skin looks much better but it is too noisy.
Case closed.

closed 2019-11-12 by PLAyerNos