bigtreetech/BIGTREETECH-SKR-mini-E3 #87: Possible fix to open the RCT chip up to 512kB

brew99, 2019-11-06:
Not really a BUG, but a possible improvement to the limited storage of the RCT chip used on the SKR mini E3/DIP.   This is posted on Reddit, I am not the author, but I have tried the proposed fix, and preliminary findings seems to be that it works.

[https://www.reddit.com/r/ender3/comments/dsct5x/in_fact_the_flash_size_of_skr_mini_e3_v12_is_512kb/](url)

Give it a shot and let's see if we can eliminate the shortcomings of the 256kB limit

steenerson, 2019-11-07:
I now have a marlin 2.0.x fork where I have applied the minimum changes to get 512K working on a SKR E3 Mini 1.2 if anyone wants to give it a shot:

https://github.com/steenerson/Marlin_SKR_E3_Mini_12_512K/

blackasdf, 2019-11-07:
Thank you very much for this @steenerson , works like a charm both with BTT and your fork.

ManiRX100, 2019-11-08:
@steenerson @brew99 What does this mean? I am not familiar with making changes in the EEPROM memory. I have seen a You Tube video by Teaching Tech on this EEPROM changes and changed a line of code from "1024" to "2048". Is this similar to that? If the memory is increased to 512K, will this affect the file sizes used for printing with the SD card? Sorry for too many questions.

steenerson, 2019-11-08:
@ManiRX100 (sorry for the wordy response I want to clear up a few things) This expands the usable program space, because the F103RC chips used on these boards say they have 256K of flash in the datasheet, but there's actually 512K on the chip that functions perfectly fine by all reports so far. The RC chips presumably share a die with the larger chips in the F103 family and ST was probably getting good yields and decided to share the wealth by not lasering off the extra flash.

I watched the Teaching Tech video and they are actually moving the EEPROM to the last 4K of the 512K space, but they don't make the other changes to fully make 512K usable. I think perhaps they didn't realize that the changes weren't possible with a 256K chip and it worked by happy accident. 256 * 2048 = 524288 bytes = 512K

The bootloader is 28K and the EEPROM is 4K, so a true 256K chip would have a hard limit at ~224K for the program, which is already somewhat low. If you want to enable many fancy features like auto bed leveling, linear advance, S-curve accel etc, you have to be careful to stay within 224K which means limiting LCD menus and things like that.

With just the EEPROM moved to the end as Teaching Tech has done, it seems it will still compile and load up to a 256K program (I haven't tested this very well), which added to the bootloader and EEPROM is up to 288K of flash used on the chip. This isn't very limiting at the moment - you have to enable almost every option to be limited by this.

The other couple changes I made allow full use of the 512K space, so theoretically up to a ~480K program should be possible. It's not necessary for most people right now, but it means you can enable whatever you want for years to come before marlin adds enough features to start hitting this limit. I've tested enabling everything to create a 260K program and it works without issues.

It should not affect printing from the SD card or gcode file sizes or anything like that affecting the normal printing workflow. It's just how much space is available to store the firmware. Hope this makes sense, ask away if you have more questions. I'm planning on adding a FAQ page so hearing these helps me out.

ManiRX100, 2019-11-09:
@steenerson Thanks for the detailed explanation. I understood how the EEPROM expansion works now. I checked your Marlin 2.0 fork and found these changes :
1. #define EEPROM_START_ADDRESS uint32(0x8000000 + 512 * 1024 - 2 * EEPROM_PAGE_SIZE)
2. rom (rx)  : ORIGIN = 0x08007000, LENGTH = 512K - 28K
3. board_upload.maximum_size = 524288
Is this logic applicable to all 3D printer  boards if I want to change the EEPROM space to 512K (Given that enough flash memory is available) ?

steenerson, 2019-11-09:
@ManiRX100 Most of the time boards are already setup to use the max available space, this is a special case where  the chips have more flash than specified. But some other chips in the STM32F103 series have doubled flash as well, so it's probably worth checking if you're working with an STM32 chip.

Every board will be setup different but you have to change the max size in platformio.ini so it will compile larger programs, then you need to find anything that references the end of the flash space. In this case they had it in the linker script, and also an emulated EEPROM in the last 4K of the 256K flash space that had to be moved to the end of 512K.

ManiRX100, 2019-11-09:
@steenerson Thank you very much. Let me flash the firmware and update the result here.

LemaruX, 2019-11-10:
Anyone hast any success with this on a v1.0 board rather than the new v1.2?

ManiRX100, 2019-11-10:
@steenerson The firmware flashed fine. Tried a 1 hour print. Print was good, but the nozzle temperature display always shows above 50 degrees even though there is no heating ON. It was working well with the previous firmware I was using.

brew99, 2019-11-12:
> Anyone had any success with this on a v1.0 board rather than the new v1.2?

I've tried many different versions of Vanilla, BTT, and @steenerson , but am not having any luck getting it to load a firmware file greater than 230kB.   It all compiles fine, but it's as if the bootloader on the v1.0 board is aborting during the beginning of a load, and just defaults back to regular start-up screen (i.e. no delay of about 8 sec for firmware load).   The bin file changes to .cur, but it doesn't actually load it.

Out of curiosity, I even tried compiling and loading the Steenerson's v1.2 w/ bltouch, and it gives the same outcome for me on the V1.0 board.

I wonder if the bootloaders are different between the two versions of boards?

steenerson, 2019-11-13:
@brew99 it sounds like there's a pretty good chance it is the bootloader locking out firmware over a certain size. If you get an ST Link you can dump the full flash contents which would backup the stock bootloader+firmware, then try replacing it with the bootloader from a 1.2 board and see if it works. If not you can put the original one back.

There's cheap versions on aliexpress for a couple bucks that work: https://www.aliexpress.com/item/32792513237.html

pmjdebruijn, 2019-11-18:
This is now available in upstream Marlin:
https://github.com/MarlinFirmware/Marlin/commit/0f57818f2decca8340fd5fc86f9ea7b792d9d802

closed 2019-11-23 by brew99