bigtreetech/BIGTREETECH-SKR-mini-E3 #88: OCTOPRINT Error: Failed to autodetect serial port, please set it manually.

cappy175, 2019-11-07:
I am having trouble getting it to connect to my SKR 1.2 mini e3. 
Error: Failed to autodetect serial port, please set it manually.
https://www.dropbox.com/s/0m18vo83dcqto16/octoprint.log?dl=0

codedmind, 2019-11-08:
Same issue

not4unco, 2019-11-11:
I'm in the same boat after enabling BLT and runout sensor.

Rushmonk, 2019-11-15:
Do you guys have a Webcam connected? Have you tried to switch USB Ports, whats the baudrate you've picked?

not4unco, 2019-11-17:
No webcam connected here and I've tried all 4 USB ports on my RPi and both on my PC. Baud rate is set to 115200.

JiraiyaK, 2019-11-25:
Same issue here, and downgrading my platform IO IDE this way bricked my python install....

iMattmax, 2020-05-31:
I have the same issue and tried different solution, nut nothing helped. So what can I do?

arnonel, 2020-06-17:
Same issue