bigtreetech/BIGTREETECH-SKR-mini-E3 #90: Unable to compile firmware

ck0park, 2019-11-08:
Hi,

I have been trying to install titan extruder, which needs recompiling firmware.
I followed exact instruction on readme, and got compile error.
Below is the error messages I got.
Tried both SKR's firmware source code, as well as modifying original Marlin 2.0.x bugfix, both giving me the same error: cannot find Conditionals_LCD.h
Help will be appreciated.

Thank you

```

> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\CK\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing LiquidTWI2
Using cache: C:\Users\CK\.platformio\.cache\85\8b02db3fff9254d346da3ee4646b5f85
LiquidTWI2 @ 0.0.0 has been successfully installed!
LibraryManager: Installing Arduino-L6470
Using cache: C:\Users\CK\.platformio\.cache\e0\a85bf8ed5b8232b743f62637d1e2c2e0
Arduino-L6470 @ 0.7.0 has been successfully installed!
LibraryManager: Installing SailfishLCD
Using cache: C:\Users\CK\.platformio\.cache\e1\f7005198ad8e3a4b619768bbbcd33ae1
SailfishLCD @ 0.0.0 has been successfully installed!
LibraryManager: Installing SailfishRGB_LED
Using cache: C:\Users\CK\.platformio\.cache\ae\92268909fe23d8d88d696e4077aa46ae
SailfishRGB_LED @ 0.0.0 has been successfully installed!
LibraryManager: Installing SlowSoftI2CMaster
Using cache: C:\Users\CK\.platformio\.cache\57\ba98e6910d02720ff6cc39cb62b43957
SlowSoftI2CMaster @ 0.0.0 has been successfully installed!
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5 #58c47ee
|-- <Adafruit NeoPixel> 1.2.4 #b72627a
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <STM32ADC> 1.0
|-- <USBComposite for STM32F1> 0.91
|-- <EEPROM>
|-- <Wire> 1.0
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Print.cpp.o
[... 158 lines cut ...]
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
========================================================================================================================================== 1 failed, 0 succeeded in 00:00:18.497 ========================================================================================================================================== 
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

```

swilkens, 2019-11-10:
I don't think you have to re-compile the firmware. You will have to change the steps / mm value, but that can be done live by using M92.

http://marlinfw.org/docs/gcode/M092.html

WHy do you need to re-compile for this?

Hobhub, 2019-11-10:
I had similar issues which I resolved by switching to vscode and making the documented changes to standard Marlin 2.0