bigtreetech/BIGTREETECH-SKR-mini-E3 #92: Buzzing sound from stepper motor after connecting USB

wakkaow, 2019-11-08:
My SKR Mini E3 board is working fine by it self.
But when ever I am trying to connect with my Mac machine the stepper motor starts buzzing sound and Board stops working completely. Only a hard reset will solve the issue.

Due to this I am not able to connect to my slicer at all.

kaeltaz, 2019-11-08:
has the same problem

steenerson, 2019-11-08:
When I power my printer up while it's plugged in to a windows machine, I get an oscillating whining sound after a few seconds and the interface freezes up. It makes the whole frame ring so it's hard to tell if it's just coming from one stepper or multiple/all of them. 

When I power it up with USB unplugged and then hot plug the cable to Windows it works normally and I can connect with pronterface and everything seems to be working.

When I plug it into a pi running octopi (Version 0.16.0, running on Raspberry Pi 3 Model B Rev 1.2 with octoprint 1.3.10) it doesn't cause the issue, no matter which order they are powered on.

@wakkaow you might try plugging the USB cable after powering up the printer as a workaround

wakkaow, 2019-11-09:
@steenerson for me in Both the cases plugging after power up. or later causes the same issue.
It looks like a over peer issue. can't say definitively.

chechilll, 2019-11-10:
Same problem. Skr mini e3 with 2209 and USB plugged to a raspberry pi with octoprint. USB cable without 5v (power cable cut)

AlanWhereAreYou, 2019-12-26:
Same problem. Noise occurs regardless of what order I power on/connect.

AlanWhereAreYou, 2019-12-27:
Ok @wakkaow, @steenerson , @chechilll - for me, the fix was actually installing the Maple USB drivers necessary to talk to to the STM32.

The drivers are in the BTT repo, but they are unsigned so they are hard to install in Win10+.

Instead I downloaded from this repo: https://github.com/rogerclarkmelbourne/Arduino_STM32

...and in \drivers\win there is a .bat file that registers them correctly. 

Once I had the right driver installed the printer stopped acting weird when I connected USB, and I was able to send gcode commands to it.