bigtreetech/BIGTREETECH-SKR-mini-E3 #93: Compiled Marlin does not work on Ender 5

DrTobogganMD, 2019-11-08:
Hi! I have the 1.2 board installed to my Ender 5, the stock firmware loads up perfect (and I can re-flash to the included .bin without any issues). However, when I go to change the settings to ender 5 specific ones and compile, the BIN file will flash, but the screen stays blue.

From what I can tell, this is an issue with Windows and PlatformIO. Is there anyone who has successfully built a BIN that works for an Ender 5?

kaeltaz, 2019-11-08:
has the same problem

bojanpotocnik, 2019-11-08:
Please check the MarlinFirmware/Marlin#15767 for solution.

DrTobogganMD, 2019-11-09:
It looks like the PR for that fix was closed without merge :(

Kecajus, 2019-11-13:
i have this same issue somehow i managed to compile firmware without a fail. Then during upload it displays:
_Uploading .pio\build\STM32F103RC_bigtree\firmware.elf
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
 
hla_swd
Error: open failed
in procedure 'program'_
So i put it on SD card manually and when restarting printer screen stays blue. how did you solve that issue?

```

DrTobogganMD, 2019-11-13:
Issue is fixed with latest Marlin download, closing issue.

@Kecajus I did not have that issue.  I did a fresh VSCode and PIO install, loaded the latest Marlin, and made the BTT changes described in the repo here and it compiled fine for me. Flashed without issue, even with a custom boot image.

closed 2019-11-13 by DrTobogganMD