bigtreetech/BIGTREETECH-SKR-mini-E3 #95: Unable to install new firmware.

Kikerumbo, 2019-11-09:
Thanks for the help in advance.

I'm trying to use TeachingTech's Marlin version (with BLTouch enabled). After a few hours reading because I could compile it (VS Code and PlatformIO) I finally got it done, copied it to the SD card (firmware.BIN), turned on the printer, the red LED blinked rapidly, then stopped.

The LCD of the printer turns on but shows nothing and the firmware.BIN in the SD card didn't change to firmware.CUR.

I tried other firmwares, but they do not compile or do the same.

Timacaro, 2019-11-09:
Same here. Anyone have a good compiled version or is this something else?

codedmind, 2019-11-09:
I use this repo (is marlin but with the lcd fix) the follow readme guide using marlin oficial and can compile and the lcd works

kcherry497, 2019-11-09:
If I used firmware.bin from the project mine boots, but when I compile using VStudio with PlatformIO LCD stays on solid.  Also tried to connect to USB with Octopi to see if the issue was only with LCD, but Octopi doesn't connect.

Kikerumbo, 2019-11-09:
> I use this repo https://github.com/ivankravets/Marlin/tree/bugfix-2.0.x (is marlin but with the lcd fix) then follow readme guide using marlin oficial and can compile and the lcd works

Thanks.

The LCD works, but nothing else is. I don't I'm going to be able to configure a stock Marlin to work with my Ender 3. Motors are homing to the wrong side, steps per mm is bonkers... etc.

I think I'll have to wait until someone does a fix. I know nothing (and I'm not even Jon Snow) about programming and copy/paste code isn't enough for this.

Thanks again.

steenerson, 2019-11-10:
@Kikerumbo I've applied the LCD fix patch to the bugfix-2.0.x-bltouch branch on my repo, give it a shot and let me know.

I haven't seen the LCD issue on my end but it at least compiles and runs fine on my machine.

Kikerumbo, 2019-11-10:
> @Kikerumbo I've applied the LCD fix patch to the bugfix-2.0.x-bltouch branch on my repo, give it a shot and let me know.
> 
> I haven't seen the LCD issue on my end but it at least compiles and runs fine on my machine.

Sir, it worked like a charm. Thanks!!!

A handshake and a beer from me.

swordfish45, 2019-11-10:
Sounds like this is your problem:

https://github.com/MarlinFirmware/Marlin/issues/15767#issuecomment-551820528

It was patched in marlin 2.x.x bugfix yesterday
https://github.com/MarlinFirmware/Marlin/pull/15853

Merge to this repo in progress
https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware/pull/131

kcherry497, 2019-11-11:
I finally got my E3 working yesterday and uploaded my code here if it helps anyone.  I updated the README with my notes.

https://github.com/kcherry497/SKR-Mini-v1.2

JiraiyaK, 2019-11-22:
> I finally got my E3 working yesterday and uploaded my code here if it helps anyone. I updated the README with my notes.
> 
> https://github.com/kcherry497/SKR-Mini-v1.2

mhmmm I got your pre-compiled firmware working, but I needed to adjust for ezabl and couldn't get Atom or VSC to build it....kept coming up with..........

""In file included from .pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:46:0:
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.h:356:3: error: 'GPIO_TypeDef' does not name a type
   GPIO_TypeDef *gpioPort;       ///< Output GPIO PORT
   ^~~~~~~~~~~~
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp: In member function 'void Adafruit_NeoPixel::show()':
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1900:23: error: 'SysTick' was not declared in this scope
   uint32_t saveLoad = SysTick->LOAD, saveVal = SysTick->VAL;
                       ^~~~~~~
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1910:28: error: 'gpioPort' was not declared in this scope
       LL_GPIO_SetOutputPin(gpioPort, gpioPin);
                            ^~~~~~~~
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1910:28: note: suggested alternative: 'gpioPin'
       LL_GPIO_SetOutputPin(gpioPort, gpioPin);
                            ^~~~~~~~
                            gpioPin
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1910:7: error: 'LL_GPIO_SetOutputPin' was not declared in this scope
       LL_GPIO_SetOutputPin(gpioPort, gpioPin);
       ^~~~~~~~~~~~~~~~~~~~
.pio\libdeps\STM32F103RC_bigtree\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp:1913:7: error: 'LL_GPIO_ResetOutputPin' was not declared in this scope
       LL_GPIO_ResetOutputPin(gpioPort, gpioPin);
       ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -fmax-errors=5.
*** [.pio\build\STM32F103RC_bigtree\libc16\Adafruit NeoPixel_ID28\Adafruit_NeoPixel.cpp.o] Error 1""

catlord21, 2020-01-02:
> mhmmm I got your pre-compiled firmware working, but I needed to adjust for ezabl and couldn't get Atom or VSC to build it....kept coming up with..........
>
> ""In file included from [...]

Same error here, the adafruit librairie seems bugy on the skr firmware