bigtreetech/BIGTREETECH-SKR-mini-E3 #96: Auto bed level guide

codedmind, 2019-11-10:
Anyone with skr mini e3 and ABL kit can make a guide what is needed to change in the firmware?

I try follow the marlin but when put g29 after g28 the printer halted

lskuared, 2019-11-11:
Following. Just installed board and hot end hangs about 2 inches above bed.

Hukuma1, 2019-11-11:
Great guide by /u/OughRay: https://github.com/OughRay/Ender3Modding/tree/master/SKR%20mini%20E3%20V1.2%20%2B%20Triangle%20Labs%203Dtouch

Also it seems by default Marlin resets (???) ABL settings once you G28.

So I enabled this in Configuration.h:

/**
 * Normally G28 leaves leveling disabled on completion. Enable
 * this option to have G28 restore the prior leveling state.
 */
#define RESTORE_LEVELING_AFTER_G28

lskuared, 2019-11-11:
So you both added the new bigtreetech board and then reflashed it with the th3d version of marlin?

codedmind, 2019-11-11:
@lskuared i installed marlin no th3d... i think th3d don't support this board

Hukuma1, 2019-11-11:
https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.0.x

codedmind, 2019-11-14:
@Hukuma1 i have EZABL  not bl touch..

nevertyam, 2019-11-25:
does the th3d version work for this? Im having issues installing any inductive probe

lskuared, 2019-11-25:
No. Th3d does not support this board. They have their own E3 board.

chazzin, 2020-01-08:
I have added an SN04 to support auto-leveling on my Ender-3, driving it with just 5V. Have ordered 24V opto-isolator and plane to drive the sensor at 24v to see if I can capitalize on the higher accuracy and put the glass back on my bed. Since I am not very circuit savvy yet I am hoping for some input for the 24V alteration using the OPTO-isolator. Currently pulling +5 off SPI and connecting signal/gnd to Z-stop pins. Not yet certain of exactly this wiring should work with opto-isolator in place.
This is the opto-isolator I ordered:

5V/12V/24V 1 Channel 220V One-Way Optical Isolation Relay Driver Board Module SV-5A1R1P-N Optocoupler Controller for PLC Control Module Component

Any input or suggestions would be appreciated. (I do not want to Fry my motherboard yet. )