bigtreetech/BIGTREETECH-SKR-mini-E3 #97: LCD Screen not working

alistair12341, 2019-11-10:
Can someone please explain step by step the solution to someone that doesn't understand the environment and everything very well how to solve the issue of customized firmware affecting the LCD screen on an Ender 3

swilkens, 2019-11-10:
First, does it work with the provided firmware?

Second, did you customize the firmware of this repo? If so, post your configuration files. If you used a different firmware (e.g. upstream Marlin) - you shoud ask on their github.

alistair12341, 2019-11-10:
I used the firmware from this with just a few parameter changes to the BLT offsets:
https://www.reddit.com/comments/dojh3v
It works if I just download the .bin file from the SKR GitHub and I've been able to compile before using Vscode+platformio

cptsoviet, 2019-11-20:
Disconnect bltouch, does the board boot up?