bigtreetech/BIGTREETECH-SKR-mini-E3 #100: [QUESTION] Using TMCStepper 6.0.0

damvcoool, 2019-11-11:
Hello BTT team,

Since teemuatlut has release TMCStepper 6.0.0, with WriteOnly Feature removed
https://github.com/teemuatlut/TMCStepper/commit/63d0ef525c5d2ba49f9fb4fb4c4440df101dd7fa

Is it still needed to use the custom (and outdated) TMCStepper from BTT?

I will be testing this later today, however some official documentation update, would be useful for future buyers.

swilkens, 2019-11-11:
I think we are still tied to a custom TMCStepper as long as @bigtreetech uses a custom swserial.

https://github.com/bigtreetech/TMCStepper/commit/7fda12fd9575f893684cb4889015bf6162436950

damvcoool, 2019-11-11:
I believe they are working on that with this PR
https://github.com/MarlinFirmware/Marlin/pull/15655#issuecomment-552599689

swilkens, 2019-11-12:
https://github.com/MarlinFirmware/Marlin/pull/15875

swilkens, 2019-11-12:
> 
> 
> [MarlinFirmware/Marlin#15875](https://github.com/MarlinFirmware/Marlin/pull/15875)

F.Y.I. the PR in this lets us use the upstream TMCStepper and Marlin cleanly, worked for me. Note that this is an intermediary step, once #15655 is merged the need for an external swserial will go away completely.

swilkens, 2019-11-13:
The changes proposed in MarlinFirmware/Marlin#15875 have been merged in upstream Marlin.

We can now build Marlin-2.0.x upstream with upstream TMCStepper for this board without the need for BTT's forks.

damvcoool, 2019-11-13:
I am compiling this with my settings right now... for reference these are the settings i am using.
I Use a BLTouch, and also changed platformio.ini to use teemuatlut repository directly.

Once i get back from work i will install the bin, and test with a Benchy.
[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3841328/Configuration.zip)

damvcoool, 2019-11-13:
I think there is something wrong with my configuration...the Printer Homes, then Bed Levels, however when printing, it's fine for the first few lines and then somehow it forgets where home is, by what i mean by that is it's supposed to draw a line that's 30cm long, and when it gets to about 20cm, it starts to execute the next command, stopping the print at this point and looking at the x,y possition it seems that it's off from where it should actually be.

I can't seem to find what may have i configured wrong.

Also with the configuration i posted it seems we get the TMC Connection Error.

swilkens, 2019-11-15:
There is a small issue with the new TMCStepper version 0.6.0 that came out recently that causes connection errors.

https://github.com/teemuatlut/TMCStepper/issues/93

For now, this can be fixed by changing your `platformio.ini` file to have `TMCStepper@=0.5.2`. Then, it should work properly again.