bigtreetech/BIGTREETECH-SKR-mini-E3 #99: Many Problems with this board.

smith505, 2019-11-11:
Just like to say emailing the support team isn't much of a help. I got ignored after a few days.
Anyways. When in either mode the board is really loud and ignores my x and y stop. Before It would get loud so I'd shut it off and move the jumper turn it on then off and move it again and it would be fine. now it's just always loud.

swilkens, 2019-11-11:
Sounds like you may have a faulty board or otherwise a faulty setup.

Try this:

1. Turn on printer, do not home and make sure the endstops are not pressed
2. Connect using something like printrun / pronterface
3. Issue a `M119` and see if the endstops are triggered. If so, something in your setup is wrong.

smith505, 2019-11-11:
@swilkens Thanks for the reply. Have done this already and they aren't triggered.