bigtreetech/BIGTREETECH-SKR-mini-E3 #101: STM32F103RC_bigtree FAILED when Compiling

PigeonFX, 2019-11-12:
Followed the guide for editing Marlin 2.0 for the SKR mini E3 V1.2, but when I try and hit CTRL+ALT+B I get STM32F103RC_bigtree FAILED. 

Not sure if this matters but noticed when pasting in
` #define MOTHERBOARD BOARD_BTT_SKR_MINI_E3_V1_2 `
it all stays yellow text, and have to manually delete it to `BTT_SKR_MINI_E3` to turn it to blue text, and then manually type the _V1_2 part.

Here is the full dump:

> Executing task in folder Marlin-bugfix-2.0.x: C:\Users\Brook Laptop\.platformio\penv\Scripts\platformio.exe run <
> 
> Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
> ---------------------------------------------------------------------------------------------
> 
> Verbose mode can be enabled via `-v, --verbose` option
> CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
> PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)    
> HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
> DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
> PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
> Converting Marlin.ino
> LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
> LDF Modes: Finder ~ chain, Compatibility ~ soft
> Found 38 compatible libraries
> Scanning dependencies...
> Dependency Graph
> |-- <U8glib-HAL> 0.4.1
> |   |-- <Wire> 1.0
> |-- <LiquidCrystal> 1.3.4
> |   |-- <Wire> 1.0
> |-- <TMCStepper> 0.4.5 #58c47ee
> |-- <Adafruit NeoPixel> 1.2.4 #b72627a
> |-- <LiquidTWI2>
> |   |-- <Wire> 1.0
> |-- <Arduino-L6470> 0.7.0
> |-- <SailfishLCD>
> |-- <SailfishRGB_LED>
> |   |-- <Wire> 1.0
> |-- <SlowSoftI2CMaster>
> |-- <STM32ADC> 1.0
> |-- <USBComposite for STM32F1> 0.91
> |-- <EEPROM>
> |-- <Wire> 1.0
> Building in release mode
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\board.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start.S.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start_c.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\syscalls.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareSerial.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareTimer.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\IPAddress.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Print.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Stream.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\WString.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\avr\dtostrf.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\cxxabi-compat.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\ext_interrupts.cpp.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\hooks.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\itoa.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\adc.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\adc_f1.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\bkp_f1.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dac.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dma.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dma_f1.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exc.S.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exti.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exti_f1.c.o
> Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\flash.c.o
[... 179 lines cut ...]
> STM32F103RC_fysetc       IGNORED
> STM32F103RC_bigtree      FAILED    00:00:11.029
> STM32F103RC_bigtree_USB  IGNORED
> STM32F4                  IGNORED
> STM32F7                  IGNORED
> ARMED                    IGNORED
> STM32F103VE_longer       IGNORED
> mks_robin                IGNORED
> mks_robin_lite           IGNORED
> mks_robin_mini           IGNORED
> mks_robin_nano           IGNORED
> jgaurora_a5s_a1          IGNORED
> STM32F407VE_black        IGNORED
> BIGTREE_SKR_PRO          IGNORED
> BIGTREE_BTT002           IGNORED
> teensy31                 IGNORED
> STM32F103CB_malyan       IGNORED
> chitu_f103               IGNORED
> teensy35                 IGNORED
> esp32                    IGNORED
> fysetc_f6_13             IGNORED
> linux_native             IGNORED
> SAMD51_grandcentral_m4   IGNORED
> include_tree             IGNORED
> =========================== 1 failed, 0 succeeded in 00:00:11.029 ===========================

zerocoolak, 2019-11-12:
Getting the same problem. searching the net to no avail all day.

zerocoolak, 2019-11-13:
> Executing task: C:\Users\M&J\.platformio\penv\Scripts\platformio.exe run <

DEPRECATED! A legacy library storage `C:\Users\M&J\Documents\GitHub\Marlin\.piolibdeps` has been found in a project.       
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `C:\Users\M&J\Documents\GitHub\Marlin\.piolibdeps` folder.
More details -> http://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
---------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing TMCStepper
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\M&J\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 295:
    pkg_dir = BasePkgManager.install(
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 45:
    obj = getattr(modules[__name__], "%sClient" % type_.title())(
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 61:
    self.check_client()
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 126:
    raise UserSideException(
=============================================== [FAILED] Took 2.02 seconds =============================================== 

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
[... 100 lines cut ...]
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
========================================== 1 failed, 0 succeeded in 00:00:02.103 ==========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

zerocoolak, 2019-11-13:
Getting the same failure.

> Executing task: C:\Users\M&J\.platformio\penv\Scripts\platformio.exe run <

DEPRECATED! A legacy library storage `C:\Users\M&J\Documents\GitHub\Marlin\.piolibdeps` has been found in a project. 
Please declare project dependencies in `platformio.ini` file using `lib_deps` option and remove `C:\Users\M&J\Documents\GitHub\Marlin\.piolibdeps` folder.
More details -> http://docs.platformio.org/page/projectconf/section_env_library.html#lib-deps
Processing STM32F103RC_bigtree (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
---------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing TMCStepper
UserSideException: Please install Git client from https://git-scm.com/downloads:
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\M&J\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\M&J\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\M&J\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 295:
    pkg_dir = BasePkgManager.install(
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 483:
    vcs = VCSClientFactory.newClient(tmp_dir, url)
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 45:
    obj = getattr(modules[__name__], "%sClient" % type_.title())(
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 61:
    self.check_client()
  File "c:\users\m&j\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 126:
    raise UserSideException(
=============================================== [FAILED] Took 2.10 seconds =============================================== 

Environment              Status    Duration
-----------------------  --------  ------------
megaatmega2560           IGNORED
megaatmega1280           IGNORED
at90usb1286_cdc          IGNORED
at90usb1286_dfu          IGNORED
DUE                      IGNORED
DUE_USB                  IGNORED
DUE_debug                IGNORED
LPC1768                  IGNORED
LPC1769                  IGNORED
sanguino_atmega644p      IGNORED
sanguino_atmega1284p     IGNORED
melzi                    IGNORED
melzi_optiboot           IGNORED
rambo                    IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      FAILED    00:00:02.103
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
STM32F103CB_malyan       IGNORED
chitu_f103               IGNORED
teensy35                 IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED
========================================== 1 failed, 0 succeeded in 00:00:02.103 ==========================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

Hobhub, 2019-11-13:
I have had similar problems but this is what I have that seems to work (compiling with vscode rather than atom) I also have bltouch enabled with a heroMe mount
[Marlin-bugfix-2.0.x.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3840175/Marlin-bugfix-2.0.x.zip)

sparrowz, 2019-11-14:
I had this issue. The warnings show git is not working, so make sure that is setup properly. I needed to restart my computer.

pmjdebruijn, 2019-11-17:
The official documentation isn't really complete, and upstream Marlin has quite a few changes related to this board, I just updated my script which does everything to build a new firmware:
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2

cptsoviet, 2019-11-20:
@PigeonFX 
> Not sure if this matters but noticed when pasting in  

If you're building from the source in this repo, the MOTHERBOARD value is already set to proper one. You don't need to do this step. [See here](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/2d686726c01b5aef68cfd3571e7a575b8980030d/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/Marlin/Configuration.h#L131). 

@zerocoolak That's a **completely** different error. You're need git.

rebeluk46, 2024-01-31:
STM32F103RC_btt_maple  FAILED    00:00:35.416 how do resolve this please