bigtreetech/BIGTREETECH-SKR-mini-E3 #102: #define Speaker not working

bluegizmo83, 2019-11-13:
I'm trying to use #define Speaker with my Ender 3 Pro so that I can play audio tones when a print finished, but when I uncomment that line I get no sound at all from the Ender 3 speaker/buzzer.

How can I make this work?

swilkens, 2019-11-13:
The firmware readme specifically states the speaker is not supported and should be commented out:

https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

diabl0w, 2019-11-16:
if your using a stock ender 3, the display doesn't have a speaker, it has a buzzer. it is a hardware thing, that buzzer on the back of the LCD is a piezo buzzer or something like that. unless you have some aftermarket speaker added, the define speaker wont work cuz you dont have one

MikeTV, 2019-12-17:
Ah, that explains why my rtttl tones aren't playing at the start & end of each print.  Sad day.

> The firmware readme specifically states the speaker is not supported and should be commented out

That sentence was removed Nov 25 with a merge in of the latest 2.0.x, probably due to it no longer causing a freeze.  Tested with `#define SPEAKER` uncommented and nothing freezes, but the beeper makes no noise at all.

> If it can make different frequency tones, Marlin supports that with the SPEAKER option enabled
-- [Marlin GitHub comment](https://github.com/MarlinFirmware/Marlin/issues/9804#issuecomment-386158170)

>In Marlin 1.1.9 the M300 command would work and was able to different tones...but now its just beeping in a fixed frequency.... If somehow you are able to make the M300 commands to work in Marlin 2.0.x, be sure to let me know, because I really miss that feature.
-- [Reddit conversation](https://www.reddit.com/r/ender3/comments/e894j7/marlin_20x_guide_for_ender_3_using_skr_mini_e3_v12/faf5vce/)

qwewer0, 2020-01-28:
On the Ender 3 display the buzzer can play M300 tones with the original creality board, but with the SKR Mini E3 v1.2 (v1 too?) can't do it. If it is set to BUZZER, it will respond to the M300 commands but with a single frequency, and if it is comment it out or set it to SPEAKER then it won't make any sound to M300 commands.
I'm really sad that it is not working on this amazing board.

Tested on: Ender 3, SKR Mini E3 v1.2, Marlin 2.0.x

> if your using a stock ender 3, the display doesn't have a speaker, it has a buzzer. it is a hardware thing, that buzzer on the back of the LCD is a piezo buzzer or something like that. unless you have some aftermarket speaker added, the define speaker wont work cuz you dont have one

diabl0w, 2020-02-02:
> On the Ender 3 display the buzzer can play M300 tones with the original creality board, but with the SKR Mini E3 v1.2 (v1 too?) can't do it. If it is set to BUZZER, it will respond to the M300 [...]

what do yo mean? its a buzzer not a speaker, it will only play buzzer sounds

qwewer0, 2020-02-02:
[This](https://youtu.be/eDqj7CBiZzs) is what I mean.

diabl0w, 2020-02-02:
> [This](https://youtu.be/eDqj7CBiZzs) is what I mean.

have you tried latest Marlin? there used to be a conflict with Timers but I believe that was fixed

qwewer0, 2020-02-02:
Will try it right away.

qwewer0, 2020-02-02:
Still not working.
If I set it to SPEAKER, it wont give out any sound, and in BUFFER it only gives out a single frequency to M300, but it plays the timing right.

diabl0w, 2020-02-02:
> Still not working.
> If I set it to SPEAKER, it wont give out any sound, and in BUFFER it only gives out a single frequency to M300, but it plays the timing right.

not timing... the timers for the board... see https://github.com/MarlinFirmware/Marlin/pull/15781

qwewer0, 2020-02-02:
Hmm, so there is a solution, just still in early stages. Interesting.

qwewer0, 2020-03-20:
In the Creality example file for the Ender 3, the SPEAKER is defined, and with the original Creality board the piezo speaker/buzzer (LCD 12864 display) was able to produce tones, like [here](https://youtu.be/eDqj7CBiZzs).

But this isn't the case with the SKR Mini E3 v1.2 board:

- If the SPEAKER is defined, the piezo speaker/buzzer won't make any sound.
- If the BUZZER if defined, the piezo speaker/buzzer makes short beeps on clicking in the printer menu, and a longer beep when saving/loading setting, but It only makes a single tone/frequency with the M300 command.

So, as far as I know, the printers piezo speaker/buzzer - Ender 3 in my case - with the SKR Mini E3 v1.2, should be able to play different tones/frequencies, but it doesn't.

I tested it on the latest bugfix (2020.03.19), Ender 3, default LCD 12864 display, SKR Mini E3 v1.2 board.

qwewer0, 2020-03-25:
> The firmware readme specifically states the speaker is not supported and should be commented out:
> 
> https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

@bluegizmo83 Can you point me, where does it say it? Because I can't see it anywhere.

thisiskeithb, 2020-04-09:
> @bluegizmo83 Can you point me, where does it say it? Because I can't see it anywhere.

It was removed at some point, but is still valid: https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/commit/f5662583c7a0c1b20950004803afef349c38df83#diff-ff79b8561e71a1edce74ddf853a29a5dL32

"comment out `//#define SPEAKER`, if you don't comment out this, it will freeze when you pressed button"

diabl0w, 2020-04-09:
can someone try to set

`#ifdef STM32_HIGH_DENSITY
    #define SERVO0_TIMER_NUM 4`

in src/HAL/STM32F1/timers.h?

there is a small chance this will fix it... my board took a shit yesterday so I am waiting for a new one

diabl0w, 2020-04-09:
> > @bluegizmo83 Can you point me, where does it say it? Because I can't see it anywhere.
> 
> It was removed at some point, but is still valid: [f566258#diff-ff79b8561e71a1edce74ddf853a29a5dL32](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/commit/f5662583c7a0c1b20950004803afef349c38df83#diff-ff79b8561e71a1edce74ddf853a29a5dL32)
> 
> "comment out `//#define SPEAKER`, if you don't comment out this, it will freeze when you pressed button"

yes but it is not supported because there is an issue with the firmware or the hardware? if the former, that is the point of opening this issue

Phonicswave, 2020-04-26:
I have exactly the same problem.

I changed my original board to the skr mini e3 and flashed it with the latest marlin 2.0.

Configuration.h (default)

// SPEAKER/BUZZER
//
// If you have a speaker that can produce tones, enable it here.
// By default Marlin assumes you have a buzzer with a fixed frequency.
//
//#define SPEAKER

//
// The duration and frequency for the UI feedback sound.
// Set these to 0 to disable audio feedback in the LCD menus.
//
// Note: Test audio output with the G-Code:
//  M300 S<frequency Hz> P<duration ms>
//
//#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
//#define LCD_FEEDBACK_FREQUENCY_HZ 5000

When I enable the speaker

#define SPEAKER;
#define LCD_FEEDBACK_FREQUENCY_DURATION_MS 2
#define LCD_FEEDBACK_FREQUENCY_HZ 5000"

to play m300 songs, I no longer have any sound at all. not even in the menu.
When i leave everything on default the sound is back. I also can play longer notes (duration) but only on the same unchangeable/default frequency.

The buzzer on the stock ender 3 is definitely able to play different frequency`s/ songs, with the original board.

bigtreetech, 2020-06-04:
Hello, everyone, according to here https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/241#issuecomment-638748565  M300 should work normally now

qwewer0, 2020-06-04:
Still not working for me.

More details [here](https://github.com/MarlinFirmware/Marlin/pull/18011#issuecomment-638786230).

marknn3, 2020-07-06:
> Still not working for me.

You must manually change file Tone.cpp and changed the TONE_CHANNEL 8 to TONE_CHANNEL 4 
See #241.
This file can be located in C:\Users\\_user_\\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\tone.cpp
And of course set #define SPEAKER in Configuration.h

This works for me on a SKR Mini E3 V2.0. 

Unfortunately the speaker is **very** loud.
Therefor I made a change to Tone.cpp to have Volume support!

This will extend the functionality of the M300 to have volume control in a very simple way:
```
M300 P100 S880    ; Play 880Hz sound with default volume
M300 P100 S10880  ; Play 880Hz sound with volume level 1 (most quiet)
M300 P100 S20880  ; Play 880Hz sound with volume level 2
M300 P100 S30880  ; Play 880Hz sound with volume level 3
```
Set the default volume:
```
M300 P0 S10000  ; Set default volume to loudest (standard volume)
M300 P0 S10001  ; Set default volume to level 1 (most quiet)
M300 P0 S10002  ; Set default volume to level 2
M300 P0 S10003  ; Set default volume to level 3
```
Placing a single `M300 S10001 P0` at the start of the GCode will make all sounds much quieter.

The following will play a friendly 'ping' sound:
```
M300 S10000 P0  ; Set default volume to loudest
M300 S02640 P50 ; Play loudest sound
M300 S32640 P50 ; Play progressively quieter sounds 
M300 S22640 P50
M300 S12640 P150
```

Please let me know your thoughts on this.

Will post the my version of Tone.cpp file later. Probably to bigtreetech/Arduino_STM32 as a pull request.

JdeV987, 2020-07-07:
> Hello, everyone, according to here [#241 (comment)](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/241#issuecomment-638748565) M300 should work normally now

Confirmed, working 100% for me.

qwewer0, 2020-07-07:
By applying the change, the speaker M300 functionality is restored in exchange for BLTouch. I will stay with the BLTouch for now.

diabl0w, 2020-07-07:
> By applying the change, the speaker M300 functionality is restored in exchange for BLTouch. I will stay with the BLTouch for now.

that has always been my experience as well, i will try with the new drivers next week sometime

bucefalo2, 2020-08-01:
> > Still not working for me.
> 
> In case you missed it, you should manually change file Tone.cpp and change TONE_CHANNEL 8 into TONE_CHANNEL 4
> See #241.
> This file can be located at C:\Users\_user_\.platformio\packages\framework-arduinoststm32-maple\STM32F1\cores\maple\tone.cpp
> And of course set #define SPEAKER in Configuration.h
> 
> This works for me on the SKR Mini E3 V2.0.
> 
> Unfortunately the speaker is **very** loud.
> Therefor I modified Tone.cpp to have Volume support!
> 
> This will extend the functionality of M300 to have volume control in a very simple and backwards compatible way:
> 
> ```
> M300 P100 S880    ; Play 880Hz sound with default volume
> M300 P100 S10880  ; Play 880Hz sound with volume level 1 (most quiet)
> M300 P100 S20880  ; Play 880Hz sound with volume level 2
> M300 P100 S30880  ; Play 880Hz sound with volume level 3
> ```
> 
> Set the default volume:
> 
> ```
> M300 P0 S10000  ; Set default volume to loudest (standard volume)
> M300 P0 S10001  ; Set default volume to level 1 (most quiet)
> M300 P0 S10002  ; Set default volume to level 2
> M300 P0 S10003  ; Set default volume to level 3
> ```
> 
> Placing a single `M300 S10001 P0` at the start of the GCode will make all sounds much more quiet.
> 
> The following will play a friendly 'ping' sound:
> 
> ```
> M300 S10000 P0  ; Set default volume to loudest
> M300 S02640 P50 ; Play loudest sound
> M300 S32640 P50 ; Play progressively quieter sounds 
> M300 S22640 P50
> M300 S12640 P150
> ```
> 
> Please let me know your thoughts on this.
> 
> Will post the my version of Tone.cpp file later. Probably to bigtreetech/Arduino_STM32 as a pull request.

this part of choosing the volum is interesting but i've no idea where i can isnert those parameters... can u share/show ur tone.cpp? thank you

Shine753, 2020-11-29:
I have the exact same issue, and no Tone.cpp on my C: disk, neither any C:\Users\[user]\.platformio\ folder

sumguysr, 2020-12-11:
> 
> 
> I have the exact same issue, and no Tone.cpp on my C: disk, neither any "C:\Users\[user]\.platformio" folder.
> The only Tone.cpp I found is in the Marlin folder, and it does not contain any TONE_CHANNEL.
> Can you help ?
I bet you need to set your file explorer to show hidden files.

Here's another way to find you .platformio folder. In Visual Studio Code click the alien icon on the left. In the quick access tree click Platforms. In the PIO Home tab click ST STM32. Then click the Reveal button in the top right hand corner. This'll open /.platformio/platforms. Go up one directory and you can locate /.platformio/packages.

trengtor, 2020-12-19:
> This will extend the functionality of M300 to have volume control in a very simple and backwards compatible way:

Very interesting

trengtor, 2020-12-21:
> Will post the my version of Tone.cpp file later. Probably to bigtreetech/Arduino_STM32 as a pull request.

Where will your new file be available?

Shine753, 2020-12-23:
OK, found, thank you.

Shine753, 2020-12-23:
> I bet you need to set your file explorer to show hidden files.

Well... they are not hidden...

> Here's another way to find you .platformio folder. In Visual Studio Code click the alien icon on the left. In the quick access tree click Platforms. In the PIO Home tab click ST STM32. Then click [...]

OK, found, thank you !