bigtreetech/BIGTREETECH-SKR-mini-E3 #104: Compiling

prahjister, 2019-11-14:
Any one having issues getting a good compile. I started atom and it said a few updates were available and I did them. Now I can't get a working firmware.bin. I have a working one from a few weeks ago. Can someone tell me the version of platform.io they are using.

prahjister, 2019-11-14:
I believe i found the issue. I'll downgrade and test tomorrow.

https://www.reddit.com/r/ender3/comments/dtirv3/skr_mini_e3_firmware_psa/?utm_source=amp&utm_medium=&utm_content=post_body

pmjdebruijn, 2019-11-14:
You may want to take a peek at:
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/61

berrenger, 2019-11-14:
Latest marlin 2.0 working

Rushmonk, 2019-11-15:
Confirmed, also no need to add the -DHAVE_SW_SERIAL in build_flags anymore.
Haven't tested if the Display is now working as I'm using a Touchscreen.

prahjister, 2019-11-15:
Does this work on the new 1.2 board as well?

sp78rus, 2019-11-18:
What board is this for?

pmjdebruijn, 2019-11-19:
As mentioned in #61, my repo https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 contains a script that works with current upstream Marlin.

Timacaro, 2019-11-20:
> As mentioned in #61, my repo https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2 contains a script that works with current upstream Marlin.

Hi,
We are both Dutch but I will continue in English. I used your repo, but for some reason the Ender 3 bed gets so hot that the alarm goes of. The 70 degrees has been passed. Do you have any idea what that can be?

pmjdebruijn, 2019-11-20:
Presumably you're just printing PLA, with a 60 degree bed settings? If so, no clue, I've never experienced that. Works just fine on my Ender 3 Pro. So I'm inclined to think you're experiencing a hardware defect, that might not have been obvious earlier?

Keep in mind that I've lowered the BED_MAXTEMP to 80, so if the alarm goes off, your bed temp sensor measured 80 degrees. Marlin just allows you to configure a temperature of 10 degrees less than BED_MAXTEMP.

my355vega, 2019-12-29:
Im having an issue with compiling. I have downloaded several different builds of marlin 2.0 for the skr mini e3 1.2 and none of them are working when trying to compile in atom. Every attempt I get STM32F103RC_bigtree_NOUSB failed. I have tried starting from scratch fallowing guides and using multiple of others builds with the thought of being able to adjust a few things.

pmjdebruijn, 2019-12-29:
_NOUSB is no longer a thing in upstream Marlin:

https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/platformio.ini#L290

my355vega, 2019-12-29:
Ok that would explain why. would you be able to help with what I would be list in the default_envs? I viewed the link you posted that takes me to L290 but im now clear on the steps I should be taking. Im still in a learning curve with 2.0 and using atom.

pmjdebruijn, 2019-12-29:
I don't follow? That page lists what each option does... If you have no specific reason to choose any specific one, just use the most basic first entry: STM32F103RC_bigtree.

Also, PlatformIO works fine standalone, I'm not sure why everyone insist on using VSCode/Atom or whatever, any random text editor will work fine with standalone PlatformIO.

As mentioned earlier in this thread, take a look at my github repo for some hints.

my355vega, 2019-12-29:
Thank you Ill take a look at your github and see if I can get this working. I am using PlatformIO but its in Atom I never was aware there was a way to do it different. the issue I am having is once i received my board the software was not operating correctly and the sd card wouldn't fix it while trying to reboot it with what BTT had stock loaded and I need to change some values im using a different hotend and hotbed over stock. the only issue im coming up with is no matter if i use STM32F103RC_bigtree or any others from the list it keeps failing.
[image]

pmjdebruijn, 2019-12-29:
You already said the build was failing, that screenshot doesn't add any new information.

Nor are you communication anything else that might help, like what repo are you using? Marlin firmware upstream? or the yet again dated SKR repo?

And most important, in the screen that tells you the build failed, you likely saw stuff flashing by, and those messages will likely tell you what's actually wrong...