bigtreetech/BIGTREETECH-SKR-mini-E3 #106: SKR mini E3 USB connection issues

MDFPereira, 2019-11-14:
Hello everyone.

After trying a lot of things, anytime I try to conect my board to the PC, nothing happens. Not even a Serial/Com port, and the board is not detected.

If I flash the firmware.bin file in the repository it works.
https://ibb.co/QfC6SJy
But If I compile from the source ones provided in the repositury I loose USB connection.
https://ibb.co/PNhxBzm

My issue is, I want to change some things in firmware and since I loose USB connection I also loose the option to run Octoprint.

Can someone help me, please?
Thank you in advance.

not4unco, 2019-11-17:
I found a fix that worked for me on reddit....
[image]

MDFPereira, 2019-11-17:
Thank you not4unco.

Everytime I try this, when opening VSCode, the core is 4.1.0...
Am I doing anything wrong?

MDFPereira, 2019-11-17:
Something is different now.
After doing everything again, platformIO is not opening under VScode and Atom...

MDFPereira, 2019-11-17:
> I found a fix that worked for me on reddit....
> [image]

Solved.
Perform this steps and reboot. After that, you will have PIO 4.0.3 installed and the error no longer happens.
Board working correctly.
Thank very much, not4unco.