bigtreetech/BIGTREETECH-SKR-mini-E3 #107: Filament not coming out

fAnts121, 2019-11-15:
So I checked everything I could imagine and it doesnt come out. The thing is that if I manually move the filament, it comes out melting from the extruder.

That means that every mechanical part works. The problem appeared when I changed the motherboard and plugged in the SKR mini e3 v1.2.
Before this, I could print everything without problems.
I dont know what else to test and how I can fix this. I googled everything but couldnt find any solution. I tried slicing new pieces, default settings on simplify3d, but nothing.

Hope someone can help me here. I checked every connection and everything is fine, as I said I can control the printer from the panel, but It doesnt print when I put anything.

fantom23, 2020-11-10:
yo man. Did u solved somehow this problem? cuz i have same problem i checked everything and dunno what to do

fAnts121, 2020-11-10:
> yo man. Did u solved somehow this problem? cuz i have same problem i checked everything and dunno what to do

Hey man, i think it was a software problem or a cable left unplugged. I don't really remember the exact solution, hope this helps

fantom23, 2020-11-10:
I have the same problem and I tried really everything and nothing is helping. Wanna check board and cables tomorrow but if it was software problem u have no idea what it was? I know I'm annoying but this thing is killing me for almost a week and I'm almost done :D

fAnts121, 2020-11-10:
> I have the same problem and I tried really everything and nothing is helping. Wanna check board and cables tomorrow but if it was software problem u have no idea what it was? I know I'm annoying but this thing is killing me for almost a week and I'm almost done :D

To be honest I think it was the Z extruder that was disconnected, but I dont really remember now. Try with different softwares to slice your piece anyways!