bigtreetech/BIGTREETECH-SKR-mini-E3 #108: Power on while plugged into Pi, causing issues with X/Y stops

virall, 2019-11-16:
If I follow the below scenario my printer seems to ignore the X/Y stops and try to print way off the board.

1. Plug in USB to Octopi (Most commonly the scenario because I just shut the printer off previously and didn't unplug it).  The screen is powered at this point with the PSU turned off.

2. Turn on the PSU and let the printer boot.

3. Hit print and watch it slam into the back of your board and try to print off the far right.

I don't see mine slamming into the board in the Z direction in this case.  It seems to remember how to do that much.  Shutting the PSU off, unplugging the Octopi, then turning the PSU on and re-plugging the Octopi seems to clear up the issue.

jrob3rts, 2019-11-16:
Hi,

I had a similar issue, to resolve I put a bit of electrical tape over the 5v pin on the usb cable so the board isn’t powered from the Pi.  So far so good, I think if you look through the closed issues, this is a workaround suggested.

taclane, 2019-11-16:
> I had a similar issue, to resolve I put a bit of electrical tape over the 5v pin on the usb cable so the board isn’t powered from the Pi. So far so good, I think if you look through the closed issues, this is a workaround suggested.

Is this the issue you were referring to? #41 

The problem the above user and I have noticed is slightly different (but also resolved by taping off the USB +5V).

When you power the SKR E3 1.2 via Octoprint USB, you can still navigate around the printer menu without crashing, but it will usually throw a "TMC Connection Error" since there likely isn't enough power for the board.  But as soon as Octoprint connects, _that error message gets overwritten_ with the printer IP or some other text.

After mains power is restored to the printer via the on/off switch, it is not apparent that the TMC drivers are still errored out.  Everything will appear entirely normal until the board receives a movement instruction, and the X/Y steppers ram the end stops and grind the belts.

virall, 2019-11-17:
I'm not seeing any errors (or IP) anywhere and my display isn't haywire as that issue is showing in the image, but otherwise the behavior you've described sounds accurate.  I will add that using Auto Home from display seems to work just fine, it seems to respect the stops there but not during an actual print.

closed 2019-11-17 by virall