bigtreetech/BIGTREETECH-SKR-mini-E3 #109: Printer will not home

redesterdragon, 2019-11-16:
Hi everyone! I am very new to all of this and decided to install the BTT SKR mini E3 v1.2 board in my ender 3 pro. After install of the new board, my printer will not auto home. I think the X Y end-stops are not working. I have reverified that they are plugged into the correct place. I am not too sure what to do from here. Any help would be greatly appreciative. Thanks!

damvcoool, 2019-11-16:
What happens when you send a G28? does it move at all?
if you try to move the X and Y using the move axis options do they move?

redesterdragon, 2019-11-16:
If I use x and y move options it moves just fine and stops when it hit and
end stop. It seems when I select auto home that the motor over steps. And
makes a clicking noise. Thank you for responding!

damvcoool, 2019-11-16:
Just out of curiosity, are you using the firmware that came with the board or have you flashed a newer release?

redesterdragon, 2019-11-16:
Hi daniel!

The issue started with the stock firmware and was persistent even after
flashing the firmware with the one off github that also had the bltouch
software enabled.

damvcoool, 2019-11-16:
where do you have your End stop connected for the BLTouch, and how is the BLTouch Wired?
if you disconnect the BLTouch does the printer home in?

redesterdragon, 2019-11-16:
I just ordered the bltouch so it is not set up yet. The printer will not
home . Could it be an issue with the stepper drivers?

damvcoool, 2019-11-16:
It that was the case, it wouldn't move at all..i think
When you said you tried a different firmware did you compile it yourself? or it was a pre-made Firmware.bin file?

redesterdragon, 2019-11-16:
It was premade

damvcoool, 2019-11-16:
I have the firmware.bin file that i build yesterday and everything seems work fine.
BLTouch is working fine, and so is homing for me...if you want to give that one a try you are welcome.
however i would strongly recommend to compile you own.
the zip file contains the config files so you can check the modifications i made, and the compiled BIN.

[Configuration.zip](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/files/3854788/Configuration.zip)

in short this is the way it works
Z-Endstop works as normal
BLTouch is connected to the dedicated Probe Endstop
I use the wiring from the following thread
[image]
Bed size is set to 225x225mm
Take a look on the config...if there is something you might like to enable/disable let me know I can recompile and post it.

redesterdragon, 2019-11-16:
Very awesome. I am out of town but when I get home I'll give this a shot.
Thank you so much!

damvcoool, 2019-11-16:
Welcome, let me know how it goes.

redesterdragon, 2019-11-16:
Will do have a nice day!

damvcoool, 2019-11-16:
Just for Transparency, I have uploaded the entire source here:
https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

redesterdragon, 2019-11-16:
I trust ya! And also appreciate your help, I am completely new to 3d
printing. And cant wait to get my printer up and running again!. So you are
the man! Thanks!

damvcoool, 2019-11-16:
To be honest, i am fairly new as well. I have only had my printer for about 1 year, and just last month i started compiling my own firmware and doing upgrades.
I strongly recommend watching Teaching Tech Youtube Channel

redesterdragon, 2019-11-16:
I agree with you, I have been watching teaching tech and making muse ...and
filament fridays

redesterdragon, 2019-11-17:
unfortunately, that did not work for me. same issue

damvcoool, 2019-11-17:
In that case, it may be a problem with the board... Send an email to BTT Support, or call them if you can.

harongabriel, 2020-07-25:
@redesterdragon have u solved your issue? Im having the same problem after installing the skr-mini-e3 v2. I can move all the axis manually but when trying to auto home the x axis keeps hitting like its stucked even far from the end stop. I don't know what to do. :/

redesterdragon, 2020-07-27:
In my case i had to check my cables. Be extra conscious of the terminal
connectors and where the wire should be. best thing is to unwire and start
from the top.