bigtreetech/BIGTREETECH-SKR-mini-E3 #110: Chirping sound from Z Axis

thebest07111, 2019-11-18:
Hello, I tried ugrading the boards firmware using the firmware.bin in the folder. from github. When I flash it the Z axis stepper starts making a chirping noise and keeps doing that. Also because of that marlin doesnt boot the marlin logo screen.

Also tried compiling myself but still the same issue. It did work with the firmware that was on the board when I installed it. 

Anybody knows a solution to this?

thebest07111, 2019-11-19:
Anyone this is the issue: https://imgur.com/a/5vpNQ7M

mrBarbecue, 2019-11-25:
Had the sams issue a month ago. I tried everything and something fixed it for me, but I am not  sure what. First of course try the older firmware. I also pressed the reset button on the mainboard. And when you trie to flash the board again, don't do anything after the lcd shows you the ender screen . Just wait a few seconds longer, then turn off the psu, wait again and finally turn it on.

nevertyam, 2019-11-25:
I updated the firmware from stock and didnt have issues. It was when i enabled inductive probe with features similar to TH3d, it gave me the exact chirping noise only when I would hit print from sd card

Ugars, 2019-11-28:
I got the same sound after doing manual mesh bed leveling, this also locks the printer. After shutting down and repowering it I got blue screen. I can only get it to work after renaming FIRMWARE.CUR to firmware.bin and do the sdcard flashing. After that my bed leveling is up to date like nothing happened(!) but whenever I do mesh bed leveling I got the same issue again and again. 
I'm regretting to get this board, too many problems, oh yeah it is silent...

odaruc, 2020-06-01:
@thebest07111  this is what mine does, about 1 in 4 times when I start the printer. Resetting it once or twice solves the issue..

Has anyone figured out a solution? I'm using the latest bugfix branch and it's still doing it.