bigtreetech/BIGTREETECH-SKR-mini-E3 #111: Can not update firmware... Get blue screen. Solid red led no flashing leds

sp78rus, 2019-11-18:
Even with existing (installed) firmware I can't access SD card. Skr mini e3 v1.2
I was able to flush firmware before and now it wont do it no metter what I tried
Please help

bojanpotocnik, 2019-11-18:
> Even with existing (installed) firmware I can't access SD card. 

I am not sure about that, but
> I was able to flush firmware before and now it wont do it no metter what I tried

please check first post in https://github.com/MarlinFirmware/Marlin/issues/15767 for solution.

pmjdebruijn, 2019-11-18:
and https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/61

sp78rus, 2019-11-18:
Still NO HELP

hedwards, 2019-11-19:
Same here, I see posts around the net of people allegedly fixing the issue, but none of the fixes seem to work and the explanations tend to be terrible.
It's beyond me why these boards are shipping without functioning firmware.

pmjdebruijn, 2019-11-19:
@sp78rus, if you put a firmware.bin on your SD card, and boot up your machine (and wait for a minute or three), is the file renamed to firmware.cur afterward?

@hedwards, mine (a V1.2) came with (basic) working firmware out of the box. Do you happen to have had an earlier board?

One of the most current sources of information about building your own firmware is my repo:
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2
A lot has changed very recently, as quite a bit of the specifics have been integrated in upstream Marlin in the past few days.

I have a prebuilt basic firmware available too:
https://github.com/pmjdebruijn/BIGTREETECH-SKR-mini-E3-V1.2/blob/83ba22ee6db3e60bcb190e55db3bfb999b11cfa5/firmware.bin?raw=true

hedwards, 2019-11-19:
@pmjdebruijn, it shipped the end of October.  From what I can tell, it cannot see anything on the SD card. There's an initial red light that goes off after a bit, but the board doesn't appear to do anything with the update files.

According to the manual, it's supposed to have the status light flicker, but that doesn't happen and the file itself doesn't appear to change name. Which leads me to think this is related to the OP.

pmjdebruijn, 2019-11-19:
Is your SD card formatted properly? Did you try another SD card (fleshly formatted)? Keep in mind the filename needs to be exactly firmware.bin in the SD card's root.

While I'm hardly an expert, and presuming you tried the above, it would seem your board might not have a bootloader, possibly because it wasn't successfully flashed at the end of production.

Presuming the chip itself is fine, you can probably flash it yourself using an ST-Link v2 (which clones of can be gotten fairly affordably these days), but as I don't have one myself, I can't provide more guidance on this topic.

If you're not willing to try that, you should probably contact the seller through whatever eCommerce platform you bought it from.

hedwards, 2019-11-22:
Looks like this may be a problem with gremlins. I did get it to update, the LCD is still not working, but I wound up waiting since yesterday and plugging everything back in again. I've got both the hot end light and that red light, but the status light appears to not function. One of the same cards that I had been using seems to have updated as the FIRMWARE.cur is now the right size and the firmware.bin is missing.

Hopefully, there's something in this that can help the OP. But, it appears to be very random as to whether or not firmware will update from the card.

stompascience, 2019-11-23:
Hi, I had a very similar problem where it would flash the .bin to .cur but just blue screen . I was using VSCode on Mac OS. I went back and recompiled with ATOM and it booted up without issues.I used the V1.2 repo from Bigtreetech with a few tweaks for hottend and linear advance.