bigtreetech/BIGTREETECH-SKR-mini-E3 #112: VSCode fails to compile the V1.2 configuration.h file

steverichab, 2019-11-19:
After downloading the appropriate V1.2 files in the repository I am unable to compile the configuration.h file using VSCode & PlatformIO.  As a result this compile failed

STM32F103RC_bigtree      FAILED    00:00:59.398

Compiling .pio\build\STM32F103RC_bigtree\src\src\lcd\menu\menu_cancelobject.cpp.o

Marlin\src\lcd\lcdprint.cpp: In function 'lcd_uint_t lcd_put_u8str_ind_P(const char*, uint8_t, lcd_uint_t)':

Marlin\src\lcd\lcdprint.cpp:50:52: error: 'LCD_FIRST_TOOL' was not declared in this scope
           lcd_put_wchar(ind + ((ch == '=') ? '0' : LCD_FIRST_TOOL));
                                                    ^~~~~~~~~~~~~~
Marlin\src\lcd\lcdprint.cpp:50:52: note: suggested alternative: 'LCD_PINS_RS'
           lcd_put_wchar(ind + ((ch == '=') ? '0' : LCD_FIRST_TOOL));
                                                    ^~~~~~~~~~~~~~
                                                    LCD_PINS_RS
Compiling .pio\build\STM32F103RC_bigtree\src\src\lcd\menu\menu_configuration.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\src\src\lcd\menu\menu_custom.cpp.o
*** [.pio\build\STM32F103RC_bigtree\src\src\lcd\lcdprint.cpp.o] Error 1

Any recomendations?

steverichab, 2019-11-19:
FYI,    Git is installed and working well.  and here is the full output from the attempt.

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (stlink) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.4.5 #58c47ee
|-- <Adafruit NeoPixel> 1.2.4 #b72627a
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <STM32ADC> 1.0
|-- <USBComposite for STM32F1> 0.91
|-- <EEPROM>
|-- <Wire> 1.0
|-- <FreeRTOS701>
|-- <Servo(STM32F1)> 1.1.2
Building in release mode
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\board.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\boards_setup.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start.S.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\start_c.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduinoVariant\wirish\syscalls.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareSerial.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\HardwareTimer.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\IPAddress.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Print.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\Stream.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\WString.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\avr\dtostrf.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\cxxabi-compat.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\ext_interrupts.cpp.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\hooks.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\itoa.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\adc.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\adc_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\bkp_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dac.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dma.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\dma_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exc.S.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exti.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\exti_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\flash.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\fsmc_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\gpio.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\gpio_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\i2c.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\i2c_f1.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\iwdg.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\nvic.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\pwr.c.o
Compiling .pio\build\STM32F103RC_bigtree\FrameworkArduino\libmaple\rcc.c.o
[... 397 lines cut ...]
sanguino_atmega1284p     IGNORED
STM32F103RE              IGNORED
STM32F103RC_fysetc       IGNORED
STM32F103RC_bigtree      FAILED    00:00:59.398
STM32F103RC_bigtree_USB  IGNORED
STM32F4                  IGNORED
STM32F7                  IGNORED
ARMED                    IGNORED
STM32F103VE_longer       IGNORED
mks_robin                IGNORED
mks_robin_lite           IGNORED
mks_robin_mini           IGNORED
mks_robin_nano           IGNORED
jgaurora_a5s_a1          IGNORED
STM32F407VE_black        IGNORED
BIGTREE_SKR_PRO          IGNORED
BIGTREE_BTT002           IGNORED
teensy31                 IGNORED
teensy35                 IGNORED
STM32F103CB_malyan       IGNORED
esp32                    IGNORED
fysetc_f6_13             IGNORED
linux_native             IGNORED
SAMD51_grandcentral_m4   IGNORED
include_tree             IGNORED

mit-gukasos, 2019-11-20:
What did you add to your config.h? I just successfully compiled the 1.2 in platformio on VSCode from clean repo clone and it never attempted to compile `.pio\build\STM32F103RC_bigtree\src\src\lcd\menu\menu_cancelobject.cpp.o` that's erroring out for you. Maybe you enabled incompatible feature?

steverichab, 2019-11-22:
OMG I finally found a source that would actually compile in VSCode without errors for my BTT SKR E3 Mini V1.2 , This system has BLTouch and is configured for sensorless homing.  Hope it helps and Thank You "Morningreis"
https://github.com/morningreis/Marlin-SKR-E3-Mini-1.2

stompascience, 2019-11-23:
Hi, also maybe try and compile with ATOM as I had the same problem with VSCode(MacOS). Have you tried editing for exactly what you want and get it to build ok in VSCode?

damvcoool, 2019-11-27:
You can also give one a try ;)
https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

TheGadge, 2019-11-28:
For anyone still stuck in this blue screen loop, just downgrade platformio and it’s core in VSC.

goos766, 2019-11-29:
> For anyone still stuck in this blue screen loop, just downgrade platformio and it’s core in VSC.

 just downgrade platformio? How to do it? Can you write a short description? Please

steverichab, 2019-11-30:
Thanks everyone!  I now have two versions that will compile for the BTT SKR Mini E3 V1.2 using VSCode & Marlin Bugfix 2.0x . It appears that my next challenge is to get it to up load to the board.
When I try to load either compiled version here is the error any help is appreciated:

CURRENT: upload_protocol = stlink
Uploading .pio\build\STM32F103RC_bigtree\firmware.elf
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1

hla_swd
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked

*** [upload] Error 1
======================================================================================================== [FAILED] Took 10.10 seconds ========================================================================================================

steverichab, 2019-12-03:
Thank you morningreis ! the code compiles in PIO and the next challenge will be to move that firmware to the BTT SKR E3 mini

closed 2019-12-03 by steverichab