bigtreetech/BIGTREETECH-SKR-mini-E3 #113: [Question] Can this board be used with the Ender 2's original LCD?

Timmiej93, 2019-11-19:
I'm looking for a replacement board for my Ender 2, since the original pooped its pants. This board has been recommended to me a few times now, as a silent and powerful option, but nobody seems to be able to conclusively tell me if it's fully compatible with the Ender 2 without replacing any parts. Can somebody here help me out more conclusively?

Justice110, 2019-11-22:
The LCD from the Ender 2 and 3 should be the same. From the looks of the boards there isn't a reason that it wouldn't be compatible

cptsoviet, 2019-11-23:
There was a person on this issue tracker trying to get ender2 screen to work but he was not able to get it to work because the way interfaces comminate is different. Worst case scenario you could probably find one real cheap compatible screen online or spend a bit more and upgrade to a small touch screen kit.

Timmiej93, 2019-11-29:
Well, this is kinda the issue I've been having. One person says it should work fine, the other says they know somebody was having issues. I guess I'll find out myself, since I ordered the board. If I remember, I'll update this post with a definitive answer.

Timmiej93, 2019-12-12:
Finally, I got a definitive answer, and actually managed to get it working. So yes, this board can be used with the original Ender2 screen, without any hardware modifications!

All credit goes to frap129 on Reddit/GitHub. [Comment in question](https://www.reddit.com/r/BIGTREETECH/comments/e89nu3/whats_the_cheapest_display_that_works_with_the/fagyezd?utm_source=share&utm_medium=web2x).

A pre-built firmware file is available [here](https://github.com/Timmiej93/Marlin/releases/tag/v1)

closed 2019-12-12 by Timmiej93