bigtreetech/BIGTREETECH-SKR-mini-E3 #115: Boot loader download and installation

hedwards, 2019-11-20:
I take it that this is a bootloader problem due to the SKR 1.2 board not even trying to read the SD card. I've already tried using several different SD cards and several different firmware flavors. I had initially thought it was related to all the firmware problems people were having.

Is there a particular file I should be using for this, or does the normal firmware for the SKR already include bootloader functionatlity? I assume that the process here is similar to flashing the stock motherboard.

hedwards, 2019-11-22:
This appears to be moot. I plugged all the wires back in and this time it appears that an update was applied. I still need to track down a firmware that fixes the LCD, but at least it appears that firmware is installing. Even if completely inexplicably.

cptsoviet, 2019-11-22:
How big is your firmware.bin in KB?

hedwards, 2019-11-22:
It's 211Kb and all of the ones I've tired are about the same. It seems to be more or less completely unpredictable when or if it will take an updated file and even when it does, the screen remains inoperable. I'm in contact with support, so hopefully something will be done.

cptsoviet, 2019-11-22:
The board already comes pre-flashed, has the screen worked for you the very
first time you powered it on?

Is the screen original ender 3?

Do you have bltouch connected?

hedwards, 2019-11-22:
It never worked, when I first installed it, it didn't even light the back light. I was eventually able to get the backlight to come on, but at this point, the only sign that the LCD is connected is that it will flicker a bit.

Yes, this is the original LCD for the ender 3 and yes I have a bltouch connected.

brew99, 2019-11-22:
double check your bltouch wiring, as different versions of bltouch swapped the grd and power wires.   It's possible that you are shorting out the 5v line and the lcd would appear to not work.

hedwards, 2019-11-23:
I've tried with and without the BLT attached as well as with and without all the other optional wires and so far it hasn't made a difference. I'm reaching the conclusion that there's something wrong with the hardware, but the lack of any available official pictures to show how the wiring should go is somewhat concerning. We're not all experts in reading diagrams and being told to just put everything in the same location is a challenge when you have to remove  all of the wires in order to have room to reattach them to the new one.

hedwards, 2019-11-27:
Apparently, my BLTouch had very wrong wiring from the factory. As in wires that were out of order so that no matter what orientation was used, it still wouldn't work. It hasn't solved my problem, but the BLTouch now self-tests and the LCD flickers. I've plugged the LCD back into my original main board and it does come on, so it doesn't appear to be damaged. 

I'm working through this with Big Tree support, but I'm still a bit surprised that there weren't proper instructions included in the box, or at least a photo or two of a properly wired board.

rcommo, 2020-09-21:
Environment: SKR mini E3 V1.2 running on an Ender 3.  Using version 2.0.5.2 of Marlin.

I was able to flash new FW up until a few days ago; now it doesn't work.  When the machine is powered up with a new version of fimrware.bin on the SD card the LCD screen stays blank for 10s of seconds.  This is what I expect.  The LCD flickers a bit and a few seconds after that the Marlin splash screen is displayed for a few seconds and finally the top level status screen.  But my version number is not the new version number, it is the old version number.

I know this because I always set the STRING_CONFIG_H_AUTHOR in Configure.h to that day's date in the form of, say, 20200917.  This has always worked fine before.

So I suspect that the something in the boot loader got busted and maybe I can save the card by flashing a new boot loader.

GadgetAngel, 2021-02-10:
here is a github repository that can help with fixing a bootloader: https://github.com/GadgetAngel/BTT_SKR_13_14_14T_SD-DFU-Bootloader