bigtreetech/BIGTREETECH-SKR-mini-E3 #117: [Request] External pin header for reset button

dburr, 2019-11-22:
For the next revision of this board, it would be nice if you could add a pin header to the board so that we can install an external reset button. I would love to be able to mount an external reset button to my case, so that I can quickly reset the board if something is wrong or if I need to update the firmware, etc.

bojanpotocnik, 2019-11-24:
This is in fact already present. Looking at the schematic, there is `K1` pushbutton which pulls the `NRST` pin low, causing the chip reset. The protection circuit (external pull-up and the capacitor) is also already present. The same pin is also connected to the `SWD` programming header.

[image]

Therefore you can mirror the onboard reset button (`K1`) functionality by connecting the external pushbutton between the `RESET` and `GND` on the `SWD` programming header.

[image]

I added 100 Ω resistor for protection, in case you would accidentally connect the button between `3.3V` and `GND` instead of `RESET` and `GND` - which is easy, you just need to turn the pinout around... Considering that you must be careful with BTR schematics (#71), it is better to be safe than sorry. Moreover, you would hardly know if the board was reset because of the `NRST` as intended or because you shortcut the `3.3V` regulator output.

dburr, 2019-12-09:
@bojanpotocnik I must have stared at the schematic for a good 10-15 minutes... I have no idea how I missed that. Thanks for pointing that out to me!

closed 2019-12-09 by dburr