bigtreetech/BIGTREETECH-SKR-mini-E3 #119: OctoPrint disconnects after running PID Auto tune M303

JWolfJr, 2019-11-23:
I recently updated my Marlin firmware after getting the lcd blue screen.  Now when trying to run M303 from my terminal tab in octoprint, after a few seconds I get disconnected. Also, the set  temperature doesn't show above the icons on my lcd screen.  I thought some how this could be related.  I also tried from Repetier Host and I can't even get any responses although it does connect.  I had no problem with this when I was first tuning for the new board when after installing into my Ender 3.  Have searched all around the forums and haven't found anything yet.

KenHoffmeyer, 2019-11-24:
Same issue here

KenHoffmeyer, 2019-11-24:
Unexpected error while reading serial port, please consult octoprint.log for details: SerialException: 'device reports readiness to read but returned no data (device disconnected or multiple access on port?)' @ comm.py:_readline:2793

JWolfJr, 2019-11-24:
Exactly what I got! I believe python files were changed to eliminate the LCD blue screen, haven't had the time to look into it too much.  Maybe someone can chime in, I'm new at posting on the forums so not to sure how to include one of the guys that worked on that fix.

JiraiyaK, 2019-11-24:
Did anyone have issues just connecting via octoprint to the board? I can't connect no matter what I do it seems.....after 2 days of tinkering with the firmware I finally have got my board flashed but USB connectivity via PC/Raspberry Pi

KenHoffmeyer, 2019-11-24:
I tried this on a bone stock Marlin 2.0 bugfix build and got the same, so I'm logging it over in the Marlin repo

JWolfJr, 2019-11-24:
When I first received my board I flashed it with my settings and the recommended changes per BTT, via VSCode on my pixelbook/linux.  Then found out my octopi wasn't acting right, all steppers running twice as fast and far as should be, after some digging around I found the SKR board doesn't like the extra 5V from the USB connection from the pi.  Then everything was perfect, awesome upgrade!  I then ran all my calibrations again extruder, flow etc ..... Had no issues using the octo print terminal, actually very convenient.  About 2 weeks ago, I went to change some firmware setting and upgrade the sd card in the SKR board and ran into the blue screen problem.  Found the work arounds for that along with the 512K fix.  Now everything is good except can't run auto tune from the octopi, auto tune runs normal from the LCD though!

meltonpieman, 2019-11-24:
> _When I first received my board I flashed it with my settings and the recommended changes per BTT, via VSCode on my pixelbook/linux. Then found out my octopi wasn't acting right, all steppers [...]

I know that you can sort the 5v problem with the strip of electrical tape method - BUT - I am sure that I read somewhere (I cannot find it again) that there is a jumper on the board to turn off the 5v coming in. Does anyone know if this is true? If so, which one?

KenHoffmeyer, 2019-11-24:
> I know that you can sort the 5v problem with the strip of electrical tape method - BUT - I am sure that I read somewhere (I cannot find it again) that there is a jumper on the board to turn off the [...]

This is off-topic from this issue.  Please create your own thread and do not clutter up a bug report with irrelevant discussions.

JWolfJr, 2019-11-24:
Didn't mean to get off topic, just putting out a little history of what I've done to get to this point. What ever can help and possibly avoid questions.

KenHoffmeyer, 2019-11-24:
> Didn't mean to get off topic, just putting out a little history of what I've done to get to this point. What ever can help and possibly avoid questions.

Yup, I understand where you were coming from.  I was referring to the reply after that from @meltonpieman which was then derailing this conversation.

ViolenceUndone, 2019-11-24:
Solution found.
https://github.com/MarlinFirmware/Marlin/issues/15993#issuecomment-557927689

JWolfJr, 2019-11-24:
Great! Thank you so much for sharing. All I need to do is change the default_envs in the platformio.ini file? Not to get off topic, but I'm assuming delete the Adafruit off the lib_ignore as well?

JWolfJr, 2019-11-25:
Never mind, I know better. Got it built and loaded just now.  Works great, thanks again for everyone's assistance.

closed 2019-11-25 by JWolfJr

bwente, 2019-12-03:
Used the Marlin update from last night. Just for reference this is what worked.
[https://github.com/MarlinFirmware/Marlin/releases/tag/2.0.0]
```
[platformio]
src_dir      = Marlin
boards_dir   = buildroot/share/PlatformIO/boards
default_envs = STM32F103RC_bigtree_512K 
```

dseven, 2020-01-17:
(copying comment from the other issue, as I'm not sure where this belongs)

Am I reading this right? The solution to the PID tuning problem is to turn off USB composite support? That's not a solution! It's like having a car that acts like the throttle pedal is stuck on full when the air conditioning is turned on, and being told by the mechanic "Well then, just don't run the air conditioning. Problem solved!".