bigtreetech/BIGTREETECH-SKR-mini-E3 #120: Won’t Auto home properly

abecaw, 2019-11-24:
I installed the SKE mini E3 V1.2 on a Ender 3 Pro. I also have a Bltouch. It worked on the original board. I followed the instructions to compile the firmware and it was successfully flashed. I installed the bltouch into the servo and into the stop connectors. I un commented out the define bltouch line and the bilinear line. When I boot up and run auto home the bed goes all the way backwards the extruder goes to the left side and contacts the switch. Then it starts to drop until I stop it by touching the bltouch pin. When it boots the bltouch goes through its test ok. The servo connectors seam loose but not enough to cause a problem. I’m considering getting a new bltouch. Help

manuforti1314, 2019-11-24:
Have you enabled 'Z Safe Homing' ? This moves the sensor to the middle of the bed for Z homing.

abecaw, 2019-11-24:
I don’t think so. I’ll give it a try and let you know. Thanks
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

abecaw, 2019-11-24:
Well I did the z safe homing and the line about X Y set at 0 0. It now auto homes correctly. When I run G29 for bed leveling it starts off of the bed on the left (facing the machine in the front). Then moves to about 1/4th if the bed then 1/2 of the bed. The bed moves a little forward and the probing continues. I stopped it because it is not going to the right side. I have bed size set to 235X235. My off sets are -42 and -15 for the nozzle.   
  

  
Maybe I should comment out the X Y set to 0 0.    Then see what happens.   
  

  
Thanks again
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

abecaw, 2019-11-24:
Everything so works. I almost got fed up and ordered a new Bltouch    
  

  
Thanks again.   
  
  
  
  
  
  
David M. Fox
 ABEC(AW)
 USN Ret.

closed 2019-11-24 by abecaw