bigtreetech/BIGTREETECH-SKR-mini-E3 #122: Board 'Bricked' After firmware update

zacharthur4223, 2019-11-25:
I was able to compile and load the latest firmware file from this repo, (red LED flashes then stops, file on card changes from .bin to .cur). However, the LCD screen remained blank and the board is now unresponsive. Is there a way to fix this short of replacing the board?

thebest07111, 2019-11-25:
I have the same issue. It first starts when inserting the usb card but after rebooting it stays blank.

bojanpotocnik, 2019-11-25:
@bigtreetech can you please update the README with information from e.g. #111 ? There is the same question more than once a week (it is natural that people just do not use search among issues...).

Also, this has been fixed in the upstream Marlin, so @zacharthur4223 and @thebest07111 try to use the latest official Marlin bugfix-2.0.x branch.

thebest07111, 2019-11-25:
> @bigtreetech can you please update the README with information from e.g. #111 ? There is the same question more than once a week (it is natural that people just do not use search among issues...). [...]

I have read it and it has something to do with platformio build. But I tough that when bigtreetech released a new firmware 10 hours ago it would include that fix?

zacharthur4223, 2019-11-25:
This was the latest Marlin bug fix as of yesterday 

Zach

kramer04, 2019-11-25:
so now the bug is fixed ?

bojanpotocnik, 2019-11-25:
> This was the latest Marlin bug fix as of yesterday

Sorry, this must then be the other issue. 8 days ago, when I last updated [my fork](https://github.com/bojanpotocnik/Marlin) (read the top of the README for changes) the issue was already gone - I think the fix was pushed upstream ~15 days ago.

thebest07111, 2019-11-26:
Very weird I keep having the same issue even when compiling default marlin firmware with only the btt changes

mmende, 2019-11-26:
I have the same issue I guess. When I flash the firmware via SD everything works fine until I turn off and on the printer, which leads to a blank screen. I just tried the latest commit in [Marlin bugfix-2.0.x (55595d4)](https://github.com/MarlinFirmware/Marlin/tree/55595d444d6969a7bff58f57959d518059acf31e) with the problem still occurring.

thebest07111, 2019-11-26:
So basiclly we have a big problem. We cant rollback because it automaticly updates and we cant compile becasue we have a blank screen.

bojanpotocnik, 2019-11-26:
> until I turn off and on the printer, which leads to a blank screen

@mmende what is the size of your compiled firmware.bin? I shall be less than 224 kB ([ 
Gummibaer's comment](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/47#issuecomment-543390714)).

Can you please try [my fork](https://github.com/bojanpotocnik/Marlin) (default or BLTouch branch) or/and the firmware.bin included my fork, so we can isolate if this is a board, compile or configuration issue? Thanks.

mmende, 2019-11-26:
It was 233 kB. I just removed SDSUPPORT (brought it down to 216 kB) which seems to fix the problem. Thanks for the tip @bojanpotocnik  Now I will only have to find out what features I can live without.

thebest07111, 2019-11-26:
What does SDsupport doe can I still use the SD card an usb or not?

mmende, 2019-11-26:
As far as I know it mainly enables printing from SD card. USB should not be effected by it.

thebest07111, 2019-11-26:
Oh okay so when you disable I cant print from sd card anymore. That kinda sucks

bojanpotocnik, 2019-11-26:
In my opinion, when you start using Octoprint you will forgot the SD card feature, so it is fine to disable it. You get slimmer menus (no SD related options in the first screen) and the space for additional features.

thebest07111, 2019-11-26:
> In my opinion, when you start using Octoprint you will forgot the SD card feature, so it is fine to disable it. You get slimmer menus (no SD related options in the first screen) and the space for additional features.

I know but for some reason whenever I connect my usb cable that has the 5V wire cut my z stepper motor starts whining.

cptsoviet, 2019-11-26:
Does firmware flashing still work with sdcard disabled?

mmende, 2019-11-26:
Yes it does

brew99, 2019-11-26:
If you find that your firmware is larger than 224kB using the env. that BTT indicates in their readme file, just change the env. to STM32F103RC_bigtree_512K and you won't need to worry about any space issues.   This seems to be working for many (including myself)

mmende, 2019-11-27:
Awesome. Thanks @brew99 ...with STM32F103RC_bigtree_512K as env the larger firmware (with SDSUPPORT) works fine even after rebooting the printer.

nmg196, 2020-05-26:
So I can't use the firmware.bin precompiled on here? I have to build my own, is that right?  (just trying to understand the solution from the comments above)