bigtreetech/BIGTREETECH-SKR-mini-E3 #123: v1.2 RGB Neopixel leds working

raymond-wortel, 2019-11-25:
After some trial and error I got my RGB leds to work.
Strip of 59 leds powered by an external wall powersupply .
https://imgur.com/a/lEBhNcu

Sharing my solution here as I spent way to much time on this...

On the WS2812B Din the normal resistor should be 300 to 500 Ohm. 
I changed it up to 12580 Ohm (several resistors in series) and now the signal is coming trough :)

Software settings :

Platformio.ini
`https://github.com/bigtreetech/Adafruit_NeoPixel`

Configuration.h
`// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  //#define NEOPIXEL_TYPE   NEO_GRB		// NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  //edit by RaymondW
  #define NEOPIXEL_TYPE (NEO_GRB + NEO_KHZ800)
  #define NEOPIXEL_PIN     PC7			// LED driving pin
  
  //Changed by RaymondW
  #define NEOPIXEL_PIXELS 59			// Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL		// Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 55		// Initial brightness (0-255)
  
  #define NEOPIXEL_STARTUP_TEST			// Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  #define NEOPIXEL_BKGD_LED_INDEX		25		// Index of the LED to use
  #define NEOPIXEL_BKGD_COLOR { 0, 0, 0, 255 }	// R, G, B, W
#endif

/**
 * Printer Event LEDs
 *
 * During printing, the LEDs will reflect the printer status:
 *
 *  - Gradually change from blue to violet as the heated bed gets to target temp
 *  - Gradually change from violet to red as the hotend gets to temperature
 *  - Change to white to illuminate work surface
 *  - Change to green once print has finished
 *  - Turn off after the print has finished and the user has pushed a button
 */
#if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
  #define PRINTER_EVENT_LEDS
#endif`

JWolfJr, 2019-11-25:
I put the 3 pixels under my hero me and are somewhat working, colors are off.  After the startup_test the first pixel is not lit.  When the hot end heats up the colors are all wrong.  Once the hotend is to temperature the first led is blue and the other two are white though.  Was putting up with it. I just need to add a high ohm resister on the dataIn line and change NEOPIXEL_TYPE? Any help is appreciated, been putting up with it but would rather it work the way its supposed to.

x0rzist, 2019-11-25:
I didn't install the LED strips yet, so I can only guess, but if your colors are off, you can try to change the NEOPIXEL_TYPE. Maybe NEO_KHZ800 or NEO_RGB.

And for the first LED behaving differently, maybe you defined NEOPIXEL_BKGD_LED_INDEX? This seems to affect only one LED.

JWolfJr, 2019-11-25:
I did try NEO_RGB, and the colors were worse. I think its weird that after power up they should be blue and the first led is off. I just tried compiling with NEOPIXEL_TYPE (NEO_GRB + NEO_KHZ800), and it failed. The output didn't like the + sign.

JWolfJr, 2019-11-25:
Also, the NEOPIXEL_BKGD_LED_INDEX is still commented out.

JWolfJr, 2019-11-25:
I'm a dummy the NEO_GRB + KHZ800 shouldn't be in parentheses! That's what I get for being lazy and copy and pasting.

raymond-w2, 2019-11-25:
> I put the 3 pixels under my hero me and are somewhat working, colors are off. After the startup_test the first pixel is not lit. When the hot end heats up the colors are all wrong. Once the hotend [...]

This was my issue also, wrong colors and partial functioning RGB leds.

JWolfJr, 2019-11-25:
Just compiled with the above changes, getting ready to test it out.  Just have to solder in the resistance change.

JWolfJr, 2019-11-25:
Thanks for posting your findings, I went with 14.4k and is pretty close.  Red is red blue is blue and so on, I think orange and yellow are slightly off.  I can live with it, I really just care about the extra light when printing.  Thanks again!

TheGadge, 2019-11-26:
Interesting! thanks for this! Did you use a cap at all? Such as adafruit recommend?

JWolfJr, 2019-11-26:
Not on this one, on all my other addressable led projects I do though. I do have a set of 4 tea lights I printed and use for a halloween scene  with a flickering sketch that I did not use any caps on.  Was on all of October with no ill affect.

demitrix, 2019-11-28:
I was able to get LEDs working with a little different route. Have you noticed any issues with inconsistent colors though? I'm curious is the resistor is the key here. I will try to get a video of the effect, but say during the warm up cycle, most pixels are purple like they should be, but a few will be yellow or red instead.

TheGadge, 2019-11-28:
> Not on this one, on all my other addressable led projects I do though. I do have a set of 4 tea lights I printed and use for a halloween scene with a flickering sketch that I did not use any caps on. Was on all of October with no ill affect.

[image]

TheGadge, 2019-11-28:
> I was able to get LEDs working with a little different route. Have you noticed any issues with inconsistent colors though? I'm curious is the resistor is the key here. I will try to get a video of [...]

“A little different route”??? Come on man, how! Lol. 
[image]

raymond-w2, 2019-11-28:
@demitrix 
It is not only the resistor or capacitor, I can get inconsistent colors if I have incorrect grounding.
The LED strip and the SKR/PSU need to share 1 common ground to get consistent colors in my setup.

I placed an extra ground wire from the SKR-PSU connection to the metal chassis and connected my 5v LED PSU ground to the chassis also.

There are multiple users mentioning inconsistent ground problems, probably pointing to a bad ground plane design.

demitrix, 2019-11-28:
Here are the changes I used when I first got my strip working.
Platformio.ini
Under [common] comment out the Adafruit Neopixel library and add
Adafruit NeoPixel=https://github.com/bigtreetech/Adafruit_NeoPixel
Under your [env:STM32F103RC_bigtree] of choice
remove Adafruit Neopixel from lib_ignore

Configuration.h
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
#define NEOPIXEL_TYPE NEO_GRB //Varies by strip
#define NEOPIXEL_PIXELS 20 // Set your length
#define NEOPIXEL_IS_SEQUENTIAL
#define NEOPIXEL_BRIGHTNESS 127 
#define NEOPIXEL_STARTUP_TEST 

The NEOPIXEL_PIN is defined in a separate file no need to define it here.

@raymond-w2 I am using the DCDC5V addon and getting 5v/gnd from the board itself. Perhaps this is an issue as well. I have a 4kOhm resistor laying around I may toss on the data in to see if it helps.

raymond-w2, 2019-11-28:
@demitrix 
Maybe redundant, but this is how I got this working.
If something changes in the future, I know I've got my details in the correct places.

Using the onboard DCDC5V is only possible with a small amount of leds, I'm using a strip of 59.
My external PSU spikes up to 1.7A when they change to full power white.

demitrix, 2019-11-28:
I only have 20 LEDs so within range. I have up for the time being. They work sometimes but colors are inconsistent, I believe it's BTTs implementation of STM32F1 in the neopixel library to be honest

demitrix, 2019-12-04:
I have made some progress in my suspicions. After some light reading about bit banging and microcontrollers and WS2812B strips I was convinced that BTTs timing was off. I just uploaded a second version my modified NeoPixel library and have much better accuracy on my LEDs. I have increased the number of nops to increase the delay between GPIOs being high and low. Some of the ranges used by BTT are far off. Really need someone with an oscilloscope to measure the data pin and confirm the timings.

`// data ONE high
// min: 550 typ: 700 max: 5,500 - Incorrect.
// min: 550 typ: 700 max: 850
GPIO_SET(pin);      
asm("nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;");
// min: 450 typ: 600 max: 5,000 - Incorrect
// min: 450 typ: 600 max: 750
GPIO_CLEAR(pin);
asm("nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop;");
      } else {
// data ZERO high
// min: 200  typ: 350 max: 500
GPIO_SET(pin);
asm("nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop;");
// data low
// min: 450 typ: 600 max: 5,000 - Very incorrect
// min: 650 typ: 800 max: 950
GPIO_CLEAR(pin);
asm("nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop; nop; nop; nop; nop; nop; nop; nop; nop; nop;"
    "nop;");`

demitrix, 2019-12-09:
So I just realized tonight when adding mesh bed leveling that my second try at this never actually flashed. The above edit to Adafruit_NeoPixel.cpp gets the RGB functionality much much closer than it was with BTTs version. anyone else using RGB feel free to try it out and see if it improves your RGB colors.

Scope666, 2020-01-05:
Hi,

Wanted to see if anyone made any progress here.  Here's the REALLY strange thing.  LED control works PERFECTLY under Klipper firmware, so there's something specific to Marlin / the Bigtree Neopixel library that's borked.

Here's my Marlin config:

```
// Support for Adafruit Neopixel LED driver
#define NEOPIXEL_LED
#if ENABLED(NEOPIXEL_LED)
  #define NEOPIXEL_TYPE   NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
  #define NEOPIXEL_PIN     PC7       // LED driving pin
  //#define NEOPIXEL2_TYPE NEOPIXEL_TYPE
  //#define NEOPIXEL2_PIN    5
  #define NEOPIXEL_PIXELS 3        // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
  #define NEOPIXEL_IS_SEQUENTIAL   // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
  #define NEOPIXEL_BRIGHTNESS 127  // Initial brightness (0-255)
  #define NEOPIXEL_STARTUP_TEST  // Cycle through colors at startup

  // Use a single Neopixel LED for static (background) lighting
  //#define NEOPIXEL_BKGD_LED_INDEX  0               // Index of the LED to use
  //#define NEOPIXEL_BKGD_COLOR { 255, 255, 255, 0 } // R, G, B, W
#endif
```

Library config in platformio.ini:

```
lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.1,<1.0.0
  #Adafruit NeoPixel
  #https://github.com/bigtreetech/Adafruit_NeoPixel
  https://github.com/demitrix/Adafruit_NeoPixel
```

Under Klipper I can address any of the 3 LED's on the hot end and get the exact color I ask for, so I know the SKR Mini E3 1.2 is capable at the hardware level.

Perfectly working Klipper config:

`[neopixel HE]                                                                                                                  pin: PC7                                                                                                                       #   The pin connected to the neopixel. This parameter must be                                                                  #   provided.                                                                                                                  chain_count: 3                                                                                                                 #   The number of Neopixel chips that are "daisy chained" to the                                                               #   provided pin. The default is 1 (which indicates only a single                                                              #   Neopixel is connected to the pin).                                                                                         color_order_GRB: True                                                                                                          #   Set the pixel order to green, red, blue. If using the WS2811 chip                                                          #   (in 800Khz mode) then set this to False. The default is True.                                                              initial_RED: 0.0                                                                                                               initial_GREEN: 0.0                                                                                                             initial_BLUE: 1.0`

wees, 2020-01-08:
I've also been trying to get my neopixel strip of 15 LEDs working. I have them plugged in directly to the neopixel port for power and data and using the same configuration.h and platform.ini as above, but my LED strip only turns on as white. I can't get the LEDs to change color, the startup test, through gcode, or through the LCD menu. Any suggestions?

Scope666, 2020-01-08:
@wees   Are you using the modified library by Demitrix?  I had WAY better results with it.  Also you might have to play with the number of LEDs in Marlin.  When I set to what I actually have it doesn't work, I had to go higher.  (I have 3, had to set to 8, 15 also works)

wees, 2020-01-08:
@Scope666 I am using Demitrix's library now. I thought that would be the golden ticket but no luck. I'll try setting a higher number of pixels. I did try a lower number thinking maybe it was a power issue and it did disable the additional LEDs, but the color is still white. Well RGB white, my strip isn't rgbw just to clarify.

Scope666, 2020-01-08:
How are you disabling the additional LEDs, aren't they all connected together in the strip?  Doing it at the software level I don't think would be enough.  My strip is also GRB only, no "W"

This one:  https://www.aliexpress.com/item/32886914735.html     (60/meter)

wees, 2020-01-08:
By disabling I mean in the software. I have a total 15 LEDs in my strip and when I set the number to 5, only 5 LEDs lit up.

demitrix, 2020-01-08:
Something is definitely off, I haven't touched this in a while been working on other things, however I noticed that each time I flash a firmware the results were different. Even with no changes to the Neopixel config, I am not sure whats going on. It looks like Klipper is using an actual timer to time the Neopixel signals, where as Marlin (well not Marlin but BTTs implementation of Neopixel) is just bitbanging the port and using nops as a "timer" to set the delay in signal. This can vary wildly in relation to the timing that the LEDs require. Hopefully one day someone with more experience (mine being none) experience with STM controllers can rewrite the library to use the proper timer or maybe Adafruit will add STM32 support someday.

Scope666, 2020-01-08:
I would suggest maybe cutting off and wiring 3 only, to replicate my setup.  You can always install them on the hot end, and then add more later to the top bar.  (my plan)

Hopefully you can get to a known working setup with the 3, and then troubleshoot from there.

Scope666, 2020-01-08:
@demitrix   For sure, Klipper definitely nailed it.  It was spot on perfect the first time I tried it, and I could individually set each LED to any color I wanted.  The Marlin guys should take a look at his code.

demitrix, 2020-01-08:
> @demitrix For sure, Klipper definitely nailed it. It was spot on perfect the first time I tried it, and I could individually set each LED to any color I wanted. The Marlin guys / BTT should take a look at his code.

Yeah I just popped over to that repo and it looks like it using an actual timer library. Maybe one day if I get bored I might dig back into this. I am in the process of moving my Ender to an enclosure and designing an electronics case for everything. So the RGB LEDs hit the backburner as I will probably light the enclosure with just white LEDs.

wees, 2020-01-08:
Had some time this morning to play around with my config. Changing to 8 LEDs didn't change anything. They all stayed white. On a whim, I changed the NEO_GRB to NEO_GRBW and my LEDs did change colors, although not the correct colors. Very strange because I'm 99% sure my LEDs do not include a W pixel. I did order the 5vDCDC board and the neopixel strip Scope666 posted above so maybe it's just my LED strip. The fact that my LEDs do respond to gcode, but don't display the correct colors leads me to believe my LED strip is out of compliance or some other standard.

demitrix, 2020-01-08:
Could possibly be the order as well. You can try NEO_RGB etc. Changing the order of R G and B until they are correct. Though most strips seem to be the default GRB.

Scope666, 2020-01-08:
If you're up to the task, you can try temporarily installing Klipper.  That library is spot on and should rule out any hardware issues.  I have the required lines posted above.

wees, 2020-01-08:
@demitrix I'll try changing the order, I didn't know you could define the order like that, thanks!
@Scope666 I've heard good things about Klipper, but I use my printer away from a Raspberry pi for ~50% of the time, can I still print from the printer's SD card if I wanted to?

Scope666, 2020-01-08:
@wees  Unfortunately no, the Pi has to be plugged straight into the printer's USB.  I was saying to just try it temporarily to see if you can get accurate colors.

Scope666, 2020-01-20:
@demitrix  I'm having the strangest issue, and I'm hoping you can help.  For some reason now when I compile Marlin with your modified Neopixel library, it completes successfully, but when I load it into the printer, it's acting like it's using the stock BTT library and not yours.  It's like something is cached or set wrong in my VSCode setup.  I still have the old firmware.bin from when I originally set it up and that works perfectly.  If I try and re-compile using the same unmodified Marlin folder, it doesn't work.  So I'm stuck using an older build if I want working LEDs.  I checked and I don't have any Global Neopixel libraries installed, I also am completely deleting the .pio folder before compiling.

Relevant parts of platformio.ini:

```
[common]
default_src_filter = +<src/*> -<src/config> -<src/HAL> +<src/HAL/shared>
extra_scripts = pre:buildroot/share/PlatformIO/scripts/common-cxxflags.py
build_flags = -fmax-errors=5 -g -D__MARLIN_FIRMWARE__ -fmerge-all-constants
lib_deps =
  LiquidCrystal
  TMCStepper@>=0.6.1,<1.0.0
  #Adafruit NeoPixel
  https://github.com/demitrix/Adafruit_NeoPixel
  U8glib-HAL=https://github.com/MarlinFirmware/U8glib-HAL/archive/bugfix.zip
  Adafruit_MAX31865=https://github.com/adafruit/Adafruit_MAX31865/archive/master.zip
  LiquidTWI2=https://github.com/lincomatic/LiquidTWI2/archive/master.zip
  Arduino-L6470=https://github.com/ameyer/Arduino-L6470/archive/0.8.0.zip
  SailfishLCD=https://github.com/mikeshub/SailfishLCD/archive/master.zip
  SailfishRGB_LED=https://github.com/mikeshub/SailfishRGB_LED/archive/master.zip
  SlowSoftI2CMaster=https://github.com/mikeshub/SlowSoftI2CMaster/archive/master.zip
```
```
[env:STM32F103RC_bigtree_512K]
platform          = ststm32
board             = genericSTM32F103RC
board_upload.maximum_size=524288
platform_packages = tool-stm32duino
build_flags       = !python Marlin/src/HAL/HAL_STM32F1/build_flags.py
  ${common.build_flags} -DDEBUG_LEVEL=0 -std=gnu++14 -DHAVE_SW_SERIAL -DSS_TIMER=4 -DSTM32_FLASH_SIZE=512
build_unflags     = -std=gnu++11
extra_scripts     = buildroot/share/PlatformIO/scripts/STM32F103RC_SKR_MINI.py
src_filter        = ${common.default_src_filter} +<src/HAL/HAL_STM32F1>
lib_deps          = ${common.lib_deps}
  SoftwareSerialM=https://github.com/FYSETC/SoftwareSerialM/archive/master.zip
lib_ignore        = SPI
monitor_speed     = 115200
```

demitrix, 2020-01-20:
@Scope666 I had the exact same experience. Seemed like every time I flashed the firmware I got a different level of LED functionality. I feel that RGB + SKR Mini E3 + Marlin is just broken until the NeoPixel library is completely overhauled. If someone could get a library like Klippers implemented we might have some better luck. I gave up for the most part and have just replaced the LEDs in my printer with plain white now that it's in an enclosure. The print status RGB just wasn't worth the hassle.

Scope666, 2020-01-20:
@bigtreetech  Are there any plans to ever fix this library?  I believe this thread is proof that the existing modified library is NOT working correctly.

wees, 2020-01-27:
Finally received my new LED strip (the one from Aliexpress linked by @Scope666 ) and now my lights work! Well mostly. They change to the correct codes via gcode and the LED LCD menu options, but only a few light up when the bed/hot end are heating up. Maybe that's intended? Once the print starts, all of them turn white. 

So it must have been something strange with the first set of LED I bought. I also put a resistor inline of data to help clean up the signal.

Scope666, 2020-01-27:
I actually gave up for now, at least until I see someone improve the modified NeoPixel library.  All I had to do was move the data line over to my Pi, I've been using this which is flawless:

NameOfTheDragon, 2020-01-28:
> Something is definitely off, I haven't touched this in a while been working on other things, however I noticed that each time I flash a firmware the results were different. Even with no changes to the Neopixel config, I am not sure whats going on. It looks like Klipper is using an actual timer to time the Neopixel signals, where as Marlin (well not Marlin but BTTs implementation of Neopixel) is just bitbanging the port and using nops as a "timer" to set the delay in signal. This can vary wildly in relation to the timing that the LEDs require. Hopefully one day someone with more experience (mine being none) experience with STM controllers can rewrite the library to use the proper timer or maybe Adafruit will add STM32 support someday.
> 
> If anyone were to have access to an oscilloscope that could measure the signals from the Neopixel pin on the SKR we'd be closer to knowing how far and which direction the timing if off by. I was just trying to make an educated guess with my changes.

I'll have a go at this soon-ish, of my cheap DSO can go fast enough to capture it. I have a roll of neopixels waiting to be fitted. I'm guessing I'll run into the same issues so I'll have a go at troubleshooting. I've had them working on an Arduino Mega so I could probably capture a working output from that for comparison.

Scope666, 2020-01-28:
That would be awesome.  Definitely look at the Klipper code for inspiration as I believe he's doing the timing the correct way.  Everything is spot on with his code on the same motherboard.

> I'll have a go at this soon-ish, if my cheap DSO can go fast enough to capture it. I have a roll of neopixels waiting to be fitted. I'm guessing I'll run into the same issues so I'll have a go at troubleshooting. I've had them working on an Arduino Mega so I could probably capture a working output from that for comparison.

NameOfTheDragon, 2020-01-28:
Leaving this here as a memorandum to self, in case I get time to work on it... https://wp.josh.com/2014/05/13/ws2812-neopixels-are-not-so-finicky-once-you-get-to-know-them/

joewmorrissey, 2020-02-12:
Can someone zip and post their marlin folder? I've made every change as listed here and still only get white lights, i know it is communicating with the board however as I can set the number of pixels to 3 and only 3 light up, or 8 and 8 light up. I'd like to compare line for line as something has got to be missing.

Scope666, 2020-02-12:
The problem is (sorry) HORRIBLE BTT modified Neopixel library, it's not even close to being correct.  I had other problems with my E3 so I upgraded to the SKR 1.4.  The CPU it uses is natively supported by the official Neopixel library so the support is perfect.

joewmorrissey, 2020-02-12:
Yeah i have a 3b setup in a dual rear case with the e3, so based upon your config, i would just move the signal to pin 24 on the GPIO on the PI and toss on the plugin

Scope666, 2020-02-13:
> Yeah i have a 3b setup in a dual rear case with the e3, so based upon your config, i would just move the signal to pin 24 on the GPIO on the PI and toss on the plugin?
> 
> I got it, it connects to pin 19 by default which is GPIO 10, and i didn't modify marlin.

Correct, the GPIO 10, the one that doesn't need Octoprint running as root, per his docs.  It's a really awesome plugin, it does print status across the top showing percentage, and you can even pick the color that goes against the white.

jwebber23, 2020-02-17:
> After some trial and error I got my RGB leds to work.
> Strip of 59 leds powered by an external wall powersupply .
> https://imgur.com/a/lEBhNcu
> 
> Sharing my solution here as I spent way to much time on this...
> 
> On the WS2812B Din the normal resistor should be 300 to 500 Ohm.
> I changed it up to 12580 Ohm (several resistors in series) and now the signal is coming trough :)
> 
> /edit
> After a bit mote tinkering best value was 14K Ohm
> 
> Software settings :
> 
> Platformio.ini
> `https://github.com/bigtreetech/Adafruit_NeoPixel`
> 
> Configuration.h
> `// Support for Adafruit Neopixel LED driver
> #define NEOPIXEL_LED
> #if ENABLED(NEOPIXEL_LED)
> //#define NEOPIXEL_TYPE NEO_GRB // NEO_GRBW / NEO_GRB - four/three channel driver type (defined in Adafruit_NeoPixel.h)
> //edit by RaymondW
> #define NEOPIXEL_TYPE (NEO_GRB + NEO_KHZ800)
> #define NEOPIXEL_PIN PC7 // LED driving pin
> 
> //Changed by RaymondW
> #define NEOPIXEL_PIXELS 59 // Number of LEDs in the strip, larger of 2 strips if 2 neopixel strips are used
> #define NEOPIXEL_IS_SEQUENTIAL // Sequential display for temperature change - LED by LED. Disable to change all LEDs at once.
> #define NEOPIXEL_BRIGHTNESS 55 // Initial brightness (0-255)
> 
> #define NEOPIXEL_STARTUP_TEST // Cycle through colors at startup
> 
> // Use a single Neopixel LED for static (background) lighting
> #define NEOPIXEL_BKGD_LED_INDEX 25 // Index of the LED to use
> #define NEOPIXEL_BKGD_COLOR { 0, 0, 0, 255 } // R, G, B, W
> #endif
> 
> /**
> 
> * Printer Event LEDs
> * During printing, the LEDs will reflect the printer status:
> * * Gradually change from blue to violet as the heated bed gets to target temp
> * * Gradually change from violet to red as the hotend gets to temperature
> * * Change to white to illuminate work surface
> * * Change to green once print has finished
> * * Turn off after the print has finished and the user has pushed a button
>     */
>     #if ANY(BLINKM, RGB_LED, RGBW_LED, PCA9632, PCA9533, NEOPIXEL_LED)
>     #define PRINTER_EVENT_LEDS
>     #endif`

I have tried this setup and it has solved my issue.  I added a 15k resistor and configured as above and it works.

The issue now is the "Click to Resume"  
[https://github.com/MarlinFirmware/Marlin/issues/16808](url)
so hopefully this is sorted and i will have some nice working LED's.

joewmorrissey, 2020-02-28:
> I have tried this setup and it has solved my issue. I added a 15k resistor and configured as above and it works.
>
> The issue now is the "Click to Resume" [...]

For those using this setup are you using genuine neo pixels from adafruit or the off brands? I have tried multiple strips in multiple lengths and can get any working but works perfect with data line from the pi. Bought a pack of 14k ohm resistors and still nothing from the e3. Really starting to aggravate the hell out of me. Trying to see if there’s a point missing like doesn’t work with actual neo pixels but does with off brand.

demitrix, 2020-02-28:
@joewmorrissey I don't think it would matter a whole lot. It's just a standard for LED control. The issue is the timing on the signal. BTTs version is wonky, mines a little closer (I think). I also have a feeling that using the noop bitbanging method will always be hard because load on the processor may change the timing causing the LEDs to not respond correctly. All in all until someone rewrites the NeoPixel library to use an actual timer like Klipper does it is probably a lost cause to try to get this working realiably. I don't have the experience nor honestly the desire to. Don't mean to be a Negative Nancy, just trying to save people time/money/frustration.

Scope666, 2020-02-28:
I gave up also, mainly because I believe a small percentage of E3's have a defect.  Myself and quite a few others have experienced a bizarre issue where the X axis runs hotter / louder than normal, and skips steps, fixed by a power cycle.  It's always X, hence the belief in a defect.

I've since switched to the SKR 1.4, which I've been a lot happier with.  As a bonus, it has perfectly working Neopixel support because the LPC1768 CPU is natively supported by Adafruit's official library.

joewmorrissey, 2020-02-28:
> I gave up also, mainly because I believe a small percentage of E3's have a defect. Myself and quite a few others have experienced a bizarre issue where the X axis runs hotter / louder than normal, [...]

Yeah I’m about to just run it from the pi as I have everything else working. Just grates at me seeing people getting it working and not able to get mine after damn near $75 bucks in crap plus hours of time lol.

Scope666, 2020-02-28:
> Yeah I’m about to just run it from the pi as I have everything else working. Just grates at me seeing people getting it working and not able to get mine after damn near $75 bucks in crap plus hours [...]

At one point I had it really close with Demitrix's version, but than just a normal Marlin update somehow screwed it up.  It's just not worth the effort, RGB Status is WAY better and every color is spot on.

demitrix, 2020-02-28:
Same issue when I was trying. Flash the same firmware  twice and get different results. That was the beginning of my giving up. Since then I just went to white LEDs the heat up cycle and everything is so short the RGB just seemed silly.

NameOfTheDragon, 2020-03-01:
> > > I gave up also, mainly because I believe a small percentage of E3's have a defect. Myself and quite a few others have experienced a bizarre issue where the X axis runs hotter / louder than normal, and skips steps, fixed by a power cycle. 

I've seen that happen if the mains power is turn off but the USB cable not unplugged (e.g. if you are using OctoPi / OctoPrint). The USB power keeps the board alive and when mains comes back it doesn't get a power on reset and that turns the board into a zombie. Check you are not back-powering the board Vcc somehow.

Scope666, 2020-03-01:
> I've seen that happen if the mains power is turn off but the USB cable not unplugged (e.g. if you are using OctoPi / OctoPrint). The USB power keeps the board alive and when mains comes back it [...]

Good theory but this wasn't it.  I learned VERY early that all E3 boards have to have the power pin of the USB cable taped off, since it doesn't have the power selector jumper like the full size SKRs do.

NameOfTheDragon, 2020-03-02:
> > Yeah i have a 3b setup in a dual rear case with the e3, so based upon your config, i would just move the signal to pin 24 on the GPIO on the PI and toss on the plugin?
> > I got it, it connects to pin 19 by default which is GPIO 10, and i didn't modify marlin.
> 
> Correct, the GPIO 10, the one that doesn't need Octoprint running as root, per his docs. It's a really awesome plugin, it does print status across the top showing percentage, and you can even pick the color that goes against the white.
> 
> Also the Rainbow Cycle effect is really cool, something you can't do with Marlin.
> 
> Sorry for the confusion, the Marlin config was if you were going to still try and go that route.

Confusingly, GPIO10 is labelled SPI0/MOSI and pin 19 on the connector, according to some pinout diagrams.

Scope666, 2020-03-02:
Yeah, pretty sure it's this one:

[image]

NameOfTheDragon, 2020-03-02:
> I actually gave up for now, at least until I see someone improve the modified NeoPixel library. All I had to do was move the data line over to my Pi, I've been using this which is flawless:
> 
> https://plugins.octoprint.org/plugins/rgb_status/

This is actually an excellent workaround. Highly recommended.

In my view, Octoprint/Octopi is far more worthwhile putting effort into if you don't have it and you solve the LED problem "en passant". The LEDs "just work" running from the Pi. I'm using a strip of 7 pixels and the Pi can power those directly, so you can just plug the strip completely into the Pi as long as you don't try to power a silly amount of pixels. Adafruit says to allow 20mA per pixel as a good rule of thumb, so my 7 pixels should only be drawing like 150mA and the official Raspberry Pi PSU has loads of headroom. You can [see this working here](https://www.facebook.com/NameOfTheDragon/videos/10158252709958833/) (the Pi is just off to the left of the printer on a stand with some breadboards, a very temporary configuration).

So I'm not putting any time into fixing my SKR Mini E3. I've given it up as a bad job.

joewmorrissey, 2020-03-02:
Yeah it is a joke. Connected up to the Pi today in 5 mins and done. Doesn't give the hot end indication and some of the features, but i mainly just wanted light for telegram to snap pics in the middle of the night.

jwebber23, 2020-03-02:
@joewmorrissey I am using a generic strip of LEDs

The key appears to be to change the NEOPIXEL_TYPE to
`#define NEOPIXEL_TYPE (NEO_GRB + NEO_KHZ800)`
and add a 15k resistor.
I am using the Bigtreetech NEOPIXEL library as per their setup info.

Please comment if you have tried this and not have it work.

Marlin 2.04.4 has the LED indication sorted now when printing from an SD card so my system is all go.....

I am also looking to try it from Octoprint but my Pi 1 is not happy with having plugins installed so that is another challenge..

[118 more replies]