bigtreetech/BIGTREETECH-SKR-mini-E3 #124: [Question] Why using STM32F103RC_bigtree_NOUSB with SKR-MINI-E3 V1.2 ?

kramer04, 2019-11-25:
on last read tells
 "If you are using the official Marlin 2.0 version, you need the following additional changes to change the default environment from megaatmega2560 to STM32F103RC_bigtree_NOUSB, env_default = STM32F103RC_bigtree_NOUSB"

This motherboard has an usb port; so i don't understand
Can someone explains this
Thanks

x0rzist, 2019-11-25:
NOUSB refers to not using USB composite ([link](https://github.com/stm32duino/Arduino_Core_STM32/pull/586)). This feature is for direct access to the SD card from the host. I think they disabled it because it uses quite a bit of PROGMEM.

It is normally not needed for normal operation but I use it to update the firmware via octoprint.

thebest07111, 2019-11-26:
But can you still use octoprint with usb connection? but just not flash with it so you need an sd card to flash?

x0rzist, 2019-11-26:
Yes, octoprint works normally even without USB composite, I use the environment STM32F103RC_bigtree_512K, where you can use more PROGMEM and USB composite is enabled.

thebest07111, 2019-11-26:
> Yes, octoprint works normally even without USB composite, I use the environment STM32F103RC_bigtree_512K, where you can use more PROGMEM and USB composite is enabled.

But for the 521K dont you need the 512K board?

x0rzist, 2019-11-26:
Somebody on [Reddit](https://old.reddit.com/r/ender3/comments/dsmf6x/512k_flash_working_on_skr_mini_e3_v12/) recently found out that the chip STM32F103RCT6 has in reality 512k flash instead of the official 256k. While not officially supported, it's working without problems for me.

kramer04, 2019-11-26:
so i use STM32F103RC_bigtree_512K and it works well
and now i've more memory, so good

Sanchowatts, 2020-09-05:
Hi everybody...
Do you mean STM32F103RC_bigtree_512K_USB or just STM32F103RC_bigtree_512K_USB ?

Thanks