bigtreetech/BIGTREETECH-SKR-mini-E3 #125: Octoprint - SKR Mini E3 - TMC Connection error

Silversurfer79, 2019-11-26:
If I boot the SKR Mini E3, it boots fine (1.2), but when I turn off the system and then plug in the Raspeberry Pi using Octoprint via USB, I get a TMC Connection error. Now the odd thing is that the XYZ axis woirk and the temps work fine, I have tried printing yet, I think it maybe fine. Any Idea why i get this error "TMC Connection error" with Octoprint?

thebest07111, 2019-11-27:
I sort of have the issue. Whenever I dont have my raspberry pi plugged in everything is fine. But whenever I plug it in I get a chirping noise from the Z axis stepper.

Ugars, 2019-11-28:
I have the same issue, I'm using bigtreetech V1.2 firmware. Normally everything is fine but if I connect my pi via USB then turn on the machine I hear a humming noise coming from the board and TMC Connection Error message. I would like to use my Pi to turn on and off the machine but because of this issue I cannot.

Silversurfer79, 2019-11-28:
Ive just noticed that I cant print either or infact if i cancel a print, the interface freezes. Could anyone tell me if this is related and where I can get thte updated BIn file? https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/2. Iv flashed the default firmware.BIN and that hasnt helped, https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2

Silversurfer79, 2019-12-01:
I have found with help from a guy on YouTube, a workaround is to unplug the Pi when booting the SKR from cold, simply plugging in the USB cable after, OctoPrint works fine and there is no Errors. 

So, this a boot issue with the SKR. With how or what it is expecting to load when the USB cable is plugged in. @Bigtreetech can you guys look at this issue please!

Silversurfer79, 2019-12-02:
Pi setup serial exceptions still
https://community.octoprint.org/t/pi-setup-serial-exceptions-still/12938/17

Silversurfer79, 2019-12-04:
@bigtreetech your responses or the lack there of, it terrible! I can’t believe how bad the support is. I hope people consider this before buying your products. If I had known, I’d go with the original Ender3 v1.1.5 board instead.

Ugars, 2019-12-04:
I wish I had know these problems before purchasing, I would not buy this crap!

Silversurfer79, 2019-12-04:
> 

> 
> 
> I wish I had know these problems before purchasing, I would not buy this crap!

https://community.octoprint.org/t/pi-setup-serial-exceptions-still/12938/17?u=silversurfer79

Silversurfer79, 2019-12-04:
These guys are useless! see that link

Ugars, 2019-12-04:
@Silversurfer79  thanks for the link, I'm also in that conversation. Don't close it, so we can see how @bigtreetech will or will not react.

Silversurfer79, 2019-12-04:
That's was a typo 😁 on my part, it's open again.

gleonard3, 2019-12-06:
I highly recommend you try this.  Solved most of my Octoprint issues immediately.  Easy enough to do or undo as the case may be.
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/41#issuecomment-541673384

Silversurfer79, 2019-12-08:
This is a work around and not a fix

harpert1, 2019-12-19:
Confirming this works.
I had a stock Ender3 with an Octopi attached.  Worked great.  I liked that setup because I could power the Ender's board without turning on the main power supply and fans- just by energizing the Octopi.

But since doing brain surgery and installing the SKR mini E3, it throws a TMC Connection Error whenever I power the E3 via the Octopi.  BUT...  If I remove the Octopi and THEN energize the E3, its works fine.  I can then plug in the Octopi again.  

Kind of a pain.  Anyway, the tape trick works.  But now I have the power the E3 with the main power supply.  I kind of like having them separate.

Thinkersbluff, 2020-02-12:
> This is a work around and not a fix

Yes it is, but I was grateful for the pointer. Hopefully, it will protect the replacement board I had to order, today.

crysxd, 2020-02-12:
You can also carefully cut open the cable 1cm wide and cut through the red cable. This is a better solution I think. I have this solution since I use octoprint because my ender 3's display would turn on otherwise with the stock board.

marcoelgordo, 2021-01-21:
2021 and this problem has not been fixed. I wouldn't have bought the SKR mini E3 if I had known that (if it ain't broken don't fix it!)