bigtreetech/BIGTREETECH-SKR-mini-E3 #128: Build keeps failing

rdagijones, 2019-11-27:
I have downloaded the zip, extracted it, opened the project in Platformio and tried to build it without making any changes.  Still the build fails.  I have attempted this on two different Windows 10 laptops and still get the identical failure.

Here is my system:
Windows 10
    Visual Studio Code 1.40.1
    Platformio Extension (Home 3.0.1 Core 4.1.0)
    Python 3.8.0 (and Python Extension installed in VSC)
    Also have installed GIT, updated Pip (Just in case)

It does not matter if I use the "master" fork from bigtreetech or any of the other forks of this fork, I get the same results.

Here is the output:
Processing STM32F103RC_bigtree_NOUSB (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
-------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
No pyvenv.cfg file
OSError: 'python Marlin/src/HAL/HAL_STM32F1/build_flags.py' exited 106:
  File "C:\Users\Raymond\appdata\local\programs\python\python38\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Raymond\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Raymond\appdata\local\programs\python\python38\lib\site-packages\platformio\builder\tools\platformio.py", line 112: 
    env.ProcessFlags(env.get("BUILD_FLAGS"))
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Raymond\appdata\local\programs\python\python38\lib\site-packages\platformio\builder\tools\platformio.py", line 202: 
    env.Append(**env.ParseFlagsExtended(flags))
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Raymond\appdata\local\programs\python\python38\lib\site-packages\platformio\builder\tools\platformio.py", line 165: 
    for key, value in env.ParseFlags(str(raw)).items():
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 807:
    do_parse(arg)
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 676:
    arg = self.backtick(arg[1:])
  File "C:\Users\Raymond\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 594:
    raise OSError("'%s' exited %d" % (command, status))
==================================================== [FAILED] Took 1.57 seconds ==================================================== 

Environment                     Status    Duration
------------------------------  --------  ------------
megaatmega2560                  IGNORED
megaatmega1280                  IGNORED
at90usb1286_cdc                 IGNORED
at90usb1286_dfu                 IGNORED
DUE                             IGNORED
DUE_USB                         IGNORED
DUE_debug                       IGNORED
LPC1768                         IGNORED
LPC1769                         IGNORED
sanguino_atmega644p             IGNORED
sanguino_atmega1284p            IGNORED
melzi                           IGNORED
melzi_optiboot                  IGNORED
rambo                           IGNORED
STM32F103RE                     IGNORED
STM32F103RC_fysetc              IGNORED
STM32F103RC_bigtree             IGNORED
STM32F103RC_bigtree_NOUSB       FAILED    00:00:01.571
STM32F103RC_bigtree_512K        IGNORED
STM32F103RC_bigtree_512K_NOUSB  IGNORED
STM32F103RE_bigtree             IGNORED
STM32F103RE_bigtree_NOUSB       IGNORED
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
=============================================== 1 failed, 0 succeeded in 00:00:01.571 ===============================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

hubris123, 2019-11-28:
I have this same problem. been having it since I got the board about 10 days ago. I tried again just now with a fresh setup and the same fail each time I try.

rdagijones, 2019-11-29:
I have found that I can compile successfully using Linux and get a Firmware.bin file.  I have not flashed it to the printer yet to verify (out of town at present) but that is my next step in testing.  Here is the rig that I put together:

- Virtual Box
- Created a virtual machine of Ubuntu 16.04 (What I had on hand.)
- Ran sudo apt update and upgrade from Terminal in Ubuntu
- Installed Atom
- Installed Clang, stm32flash and made sure Git and Pip were up to date (Python is already part of Ubuntu)
- Added the Python and Platformio extensions to Atom
-Downloaded the Marlin fork for BigTreeTech and followed the instructions on their Git site to set up platformio.ini without any other modifications
-Success

I think the problem is Windows specific.  I wonder if it has something to do with SCons and the build environment that the SKR boards are expecting.  Thoughts?

DoctorDanke, 2019-11-29:
I also just experienced this problem, using VSCode and PlatformIO.  After hours of not knowing what I was doing wrong, with a little research, I've found the fix, and the fix is irritatingly simple.  It appears that some part of the Windows environment is encountering a filename or path name limit, and all you have to do is rename the folder/folders to something shorter than "BIGTREETECH-SKR-mini-E3-master".  I went to the extreme and actually pulled the "BIGTREETECH-SKR-mini-E3-master\firmware\V1.2\Marlin-2.0.x-SKR-Mini-E3-V1.2" folder out of the main folder structure (to further reduce the path name size), renamed the "Marlin-2.0.x-SKR-Mini-E3-V1.2" folder to just "Marlin-2.0.x", and opened the "Marlin-2.0.x" folder as a project in VSCode, and it now builds and works just as expected, and doesn't fail to build anymore.

It appears in the future, BTT is going to have to reduce the large number of characters in the folder structure.

hubris123, 2019-11-29:
thank you so much that worked perfectly and built on the first try. I used the exact folders BTT had setup so the file name length was 2 times as bad.

\BIGTREETECH-SKR-mini-E3-master\BIGTREETECH-SKR-mini-E3-master\firmware\

rdagijones, 2019-12-26:
I tried shortening the filename and path and still have the same build fail with the same error.  I am still using the linux workaround.