bigtreetech/BIGTREETECH-SKR-mini-E3 #129: Question: does USB composite work for PC as of now?

brew99, 2019-11-28:
I'm trying the latest vanilla marlin (nov 27th) with the USB env, and when I plug in my PC (Win10 nov update) to the mini E3, It asks to format the drive., but if I select to do that, it fails.   Device manager shows a USB COM port, and under drives it shows a ST drive, but it is not accessible.   I've tried 2 diff SD cards.

Any help on this? (sorry don't know where else to ask

brew99, 2019-12-06:
A user on the Marlin Discord channel found a work around by disabling the SD_CHECK_AND_RETRY in configuration.h, which allowed access to SD.   Could be further complications, so use at your own risk.

closed 2019-12-06 by brew99

reloxx13, 2020-01-13:
> 
> 
> A user on the Marlin Discord channel found a work around by disabling the SD_CHECK_AND_RETRY in configuration.h, which allowed access to SD. Could be further complications, so use at your own risk.

its not a workaround, its not even enabled by default in the marlin example config = fail configuration whereever u got it enabled from and it works fine without crc check since months for me.

brew99, 2020-01-13:
Yes and that’s why I closed it on Dec 6.  I was working with that user