bigtreetech/BIGTREETECH-SKR-mini-E3 #132: Problems uploading to v1.2 from Atom

h00etn, 2019-11-30:
Hi,

I'm having trouble uploading the fw (v1.2) with Atom. Building is successful, but I get this error when I try to upload it to the connected board (on OSX). The board shows up in Atom as '/dev/cu.usbmodem141101'

What should I do/change?

_Building in release mode
Checking size .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
DATA:    [===       ]  26.4% (used 12952 bytes from 49152 bytes)
PROGRAM: [========  ]  82.3% (used 215868 bytes from 262144 bytes)
Configuring upload protocol...
AVAILABLE: blackmagic, dfu, jlink, serial, stlink
CURRENT: upload_protocol = stlink
Uploading .pio/build/STM32F103RC_bigtree_NOUSB/firmware.elf
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-15:21)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
debug_level: 1
 
hla_swd
Error: open failed
in procedure 'program'
** OpenOCD init failed **
shutdown command invoked
 
*** [upload] Error 1_

RandyatGretchen, 2020-01-18:
I'm having the exact same problem.

jignoni, 2020-02-01:
Running into this issue aswell.
Anyone found a solution ?