bigtreetech/BIGTREETECH-SKR-mini-E3 #133: Firmware.bin updated. Now fan 100% nonstop

neobikes, 2019-11-30:
just flashed with firmware.bin 2 days ago.
Now i got my fan working 100% nonstop no matter what. if it stops/prints/auto homes always 100% fan

DoctorDanke, 2019-11-30:
Make sure that your part cooling fan is plugged into the FAN0 port of your SKR Mini E3 board.  When installing the board, this is the only plug-in item that doesn't go in the same place as the stock board.  Possibly, you just didn't realize that the fan was always running at 100%.  If you do have it plugged in the correct port (FAN0 is the only speed controlled fan port), then please disregard this.

neobikes, 2019-12-02:
Extruder got 2 fans
Connected both fans mirror to old stock ender 3 board.
Both fans 100% nonstop.

I am not sure if swap that will still make both fans working 100%

ps. sorry that i didnt mentioned about BOTH runnning 100% at first i edited.

norbu15, 2019-12-04:
the reason mainboard fan running non stop is heatsink is missing on mosfet... so annoying and loud. on creality its doesn't need to run all the time because it has heat sink.

closed 2019-12-08 by neobikes