bigtreetech/BIGTREETECH-SKR-mini-E3 #134: it refuses to build

rgar2277, 2019-11-30:
macOS, VSCode, I tried compiling it but I get this output:
> Executing task in folder marlin2.0: platformio run <

Processing STM32F103RC_bigtree_NOUSB (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: tool-stm32duino 1.0.1, toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0)
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 39 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <U8glib-HAL> 0.4.1
|   |-- <Wire> 1.0
|-- <LiquidCrystal> 1.3.4
|   |-- <Wire> 1.0
|-- <TMCStepper> 0.6.1
|   |-- <SoftwareSerialM>
|-- <Adafruit NeoPixel> 1.2.4 #b72627a
|-- <LiquidTWI2>
|   |-- <Wire> 1.0
|-- <Arduino-L6470> 0.7.0
|-- <SailfishLCD>
|-- <SailfishRGB_LED>
|   |-- <Wire> 1.0
|-- <SlowSoftI2CMaster>
|-- <SoftwareSerialM>
|-- <EEPROM>
|-- <USBComposite for STM32F1> 0.91
|-- <Wire> 1.0
|-- <STM32ADC> 1.0
Building in release mode
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/board.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/boards.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/boards_setup.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/start.S.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/start_c.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduinoVariant/wirish/syscalls.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/HardwareSerial.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/HardwareTimer.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/IPAddress.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/Print.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/Stream.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/WString.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/avr/dtostrf.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/cxxabi-compat.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/ext_interrupts.cpp.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/hooks.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/itoa.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/adc.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/adc_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/bkp_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/dac.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/dma.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/dma_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/exc.S.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/exti.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/exti_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/flash.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/fsmc_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/gpio.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/gpio_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/i2c.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/i2c_f1.c.o
Compiling .pio/build/STM32F103RC_bigtree_NOUSB/FrameworkArduino/libmaple/iwdg.c.o
[... 115 lines cut ...]
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
========================================================== 1 failed, 0 succeeded in 00:00:14.371 ==========================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

buhralex, 2019-12-01:
Please redownload the sources.

> Serial port 0 does not exist

Serial port should be set to -1

neobikes, 2019-12-02:
in file
..\Marlin\configuration.h
#define SERIAL_PORT 2
#define SERIAL_PORT_2 -1

in file
..\config\examples\Creality\Ender-3\configuration.h
#define SERIAL_PORT 0
//#define SERIAL_PORT_2 -1

using VSCode
using these files from ..\config\examples\Creality\Ender-3\
same error
"Serial port 0 does not exist"

so its not updated in ..\config\examples\Creality\Ender-3\configuration.h