bigtreetech/BIGTREETECH-SKR-mini-E3 #135: Board stopped working after Bltouch install

slonkar, 2019-12-01:
Hello,
I just added Bltouch to Ender 3. I'm using firmware provided by you (firmware-bltouch-for-z-homing.bin) I am using servo and Z Stop pins to connect my Bltouch.

After the firmware is flashed (Checked info to make sure it was indeed flashed) as soon as, I plugin Bltouch wires in board LCD turns off along with power indicator red led on board. If I remove the 3 cable pin from servo pins then printers start up normally.
I have double checked the connect and order of wires on servo pin (Yellow, Brown, Red) and Z Stop Pin (Black, White).

Did I miss anything?

buhralex, 2019-12-01:
It sound like you plugged in the cables while the board still was powered. Never do that. That can fry your board.

bojanpotocnik, 2019-12-01:
> I have double checked the connect and order of wires on servo pin (Yellow, Brown, Red) and Z Stop Pin (Black, White).

That is the problem - check [this image](https://github.com/bojanpotocnik/Marlin/tree/bugfix-2.0.x_SKR-mini-E3-V1.2_BLTouch) and [this comment](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/70#issuecomment-546904235) for proper connection and explanation.

damvcoool, 2019-12-05:
Take a look at the wiring for the creality branded bltouch on my repo it may give you some idea of what to troubleshoot
https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

fnxpt, 2020-01-07:
@bojanpotocnik just to be sure the correct pins are GND/+5V/PA1 for v1.2