bigtreetech/BIGTREETECH-SKR-mini-E3 #136: Probe Offset cannot  be stored

Nikcos, 2019-12-01:
I m trying to set the offset of the Bl touch , but Skr mini E3 dont save it to memory (M851 Z-0.9  and m500), after i restart the printer, m851 is Z0.00 ,  for the moment i have put the offset value in cura's start g-code and its ok , but i wish to fix this , does anyone else having that problem ???
BTW my firmware is 217Kb , i dont think thats the problem

buhralex, 2019-12-01:
Have you turned off the Software endstop protection?
To verify you can go to the motion menu and try to move the z-axis below 0. If it's not working.
You can send a M211 S0 to disable it. Just remember to turn it back on when you've finished setting your z-offset (M211 S1)

Nikcos, 2019-12-01:
> Have you turned off the Software endstop protection?
> To verify you can go to the motion menu and try to move the z-axis below 0. If it's not working the protection is enabled.
> You can send a M211 S0 to disable it. Just remember to turn it back on when you've finished setting your z-offset (M211 S1)

I can set negative values , the problem is that these values cannot be saved , after turning the printer off and on again, the value returns to z0.00

buhralex, 2019-12-01:
I had the same problem with a self compiled firmware today.
I spend hours trying to fix this. Now I just use the precompiled firmwares from this repository. It seems to be working now.

wjones1972, 2019-12-02:
I am having the same issue with the z offset not holding after the power is turned off. I am unable to use the reconfigured firmware because I have a EZABLE instead of bltouch  looking for a solution.

smcallis77, 2019-12-02:
Sounds like this is the same as issue #127 unless you can save other settings e.g. M92 etc and its just the z offset you cannot save.

Nikcos, 2019-12-02:
I can confirm what DoctorDanke said in the other post , the values have gone after the second reboot , in the first i can say  are ok , but when i turn it off second time , values are gone to default

neobikes, 2019-12-02:
Are normal configuration settings saved?
Can you go to configuration->adv.configuration menu and change for example E-steps from 100.0 to lets say 101 and save.
power off, power on
check settings if they are saved?

I managed to compile today and my settings do get saved now.
I used Marlin official and hand tweaked settings in configuration.h based on various info found around.

goos766, 2019-12-02:
I need a little more changes than just Z offset my BLtouch probe has a different position than most users. I need to change parameters in extruder step settings or nozzle position (I have BP6 hotend with direct drive extruder with mounting plate) etc. but I still have problems compiling the file because no one is able to give a step-by-step recipe how to do it on the Windows 10 platform. most of us users must do something wrong when installing software or when choosing compile options. changes made to the last update from bigtreetech do not end only with 2 configuration files. buying from bigtreetech I was counting on some understanding for less advanced users

buhralex, 2019-12-02:
I created a fork of the current Marlin release for Ender 3 with BLTouch enabled.
[https://github.com/buhralex/Marlin](url)
There is a precompiled firmware.bin in the root directory which works for me. I haven't fully tested everything but it should be fine. I can compile it without errors and for now it seems that the settings are stored.

I use VSCode with the lates PlatformIO release.

goos766, 2019-12-02:
**buhralex** I will definitely try your version

brew99, 2019-12-04:
I'm almost certain your issue is related to [this](https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/issues/127).   Disable PRINTCOUNTER in configuration.h and it should work

damvcoool, 2019-12-05:
Take a look at the eeprom fix on my repo, this should work for you.
https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

mtaylor819, 2019-12-06:
What is the eeprom fix buddy ? I have compiled from the latest 2.0 release, so would like to incorporate the fix into my build if possible ?

damvcoool, 2019-12-06:
In my repo, the eeprom fix was to modify marlin/src/pins/stm32/skr_mini and change the location of eeprom from 1024 to 2048

However, there is a new (and better) eeprom fix
https://github.com/MarlinFirmware/Marlin/pull/16118

kamiloski88, 2020-04-02:
> Take a look at the eeprom fix on my repo, this should work for you.
> https://github.com/damvcoool/Marlin-2.0.x-SKR-Mini-E3-v1.2

I cant see this change fron 1024 to 2048 here in your repo marlin/src/pins/stm32/skr_mini

damvcoool, 2020-04-02:
@kamiloski88 this change is no longer needed since Marlin 2.0.3

kamiloski88, 2020-04-02:
> @kamiloski88 this change is no longer needed since Marlin 2.0.3

So if i want to update my marlin. I have to clone newest repo and copy/pase my config files? and it is enough?

damvcoool, 2020-04-03:
@kamiloski88 
pretty much... i would suggest to compare your config files against the newest version to make sure that all the settings still apply.

00monter00, 2020-04-16:
> PRINTCOUNTER w configuration.h

For me helps, thank you.

Darkwing69, 2020-04-30:
having same issue with newest bigtree firmware. I set the z offset and if i turn the printer back off it goes back to 0.0. I have done a M500 and it shows saved but everytime i turn it off then back on it goes to default. I have a buddy with another machine and he is having the same issues. Should i enable printercounter? its currently disabled.

00monter00, 2020-04-30:
> having same issue with newest bigtree firmware. I set the z offset and if i turn the printer back off it goes back to 0.0. I have done a M500 and it shows saved but everytime i turn it off then back on it goes to default. I have a buddy with another machine and he is having the same issues. Should i disable printcounter?

A bit funny that you had to ask. It has helped many people and will probably help you too. Faster you would disable PRINCOUNTER and upload soft than you wrote this post.
Try for the future because that's how you learn;)

Darkwing69, 2020-04-30:
I have done that before and caused more issues with with the printer than i started with. Always question then edit firmware.

leoavilagg, 2023-04-03:
Perdon si mi mensaje esta fuera de contexto, pero este post lo visite para solucionar un problema con el m851 que no podia configurar en la eeprom, tengo un sensor sn04-n y con las siguientes lineas pude habilitarlo  para editar:

en Configuration.h

#define FIX_MOUNTED_PROBE
#define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN
#define AUTO_BED_LEVELING_BILINEAR
Z_MIN_PROBE_ENDSTOP_INVERTING TRUE
#define Z_AFTER_HOMING 15