bigtreetech/BIGTREETECH-SKR-mini-E3 #137: Can't compile latest version

patzik, 2019-12-02:
I keep getting the following error when compiling the latest version.

Cloning into 'D:\SKR\.pio\libdeps\STM32F103RC_bigtree_NOUSB\_tmp_installing-u6m_xflu-package'...

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
BUG: refs/files-backend.c:2889: initial ref transaction called with existing refs
PlatformioException: VCS: Could not process command ['git', 'clone', '--recursive', '--depth', '1', 'https://github.com/bigtreetech/Adafruit_NeoPixel', 'D:\\SKR\\.pio\\libdeps\\STM32F103RC_bigtree_NOUSB\\_tmp_installing-u6m_xflu-package']:
  File "C:\users\game pc\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Game PC\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Game PC\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\users\game pc\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Game PC\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py",
line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\game pc\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\users\game pc\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Game PC\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py",
line 224:
    return self.method(*nargs, **kwargs)
  File "C:\users\game pc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py",
line 1027:
    project.install_dependencies()
  File "C:\users\game pc\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py",
line 859:
    lm.install(uri)
  File "c:\users\game pc\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 295:
    pkg_dir = BasePkgManager.install(
  File "c:\users\game pc\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "c:\users\game pc\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 484:
    assert vcs.export()
  File "c:\users\game pc\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 165:
    assert self.run_cmd(args)
  File "c:\users\game pc\.platformio\penv\lib\site-packages\platformio\vcsclient.py", line 104:
    raise PlatformioException("VCS: Could not process command %s" % e.cmd)
================================== [FAILED] Took 3.94 seconds ==================================

Environment                     Status    Duration
------------------------------  --------  ------------
megaatmega2560                  IGNORED
megaatmega1280                  IGNORED
at90usb1286_cdc                 IGNORED
at90usb1286_dfu                 IGNORED
DUE                             IGNORED
DUE_USB                         IGNORED
DUE_debug                       IGNORED
LPC1768                         IGNORED
LPC1769                         IGNORED
sanguino_atmega644p             IGNORED
sanguino_atmega1284p            IGNORED
melzi                           IGNORED
melzi_optiboot                  IGNORED
rambo                           IGNORED
STM32F103RE                     IGNORED
STM32F103RC_fysetc              IGNORED
STM32F103RC_bigtree             IGNORED
STM32F103RC_bigtree_NOUSB       FAILED    00:00:03.944
STM32F103RC_bigtree_512K        IGNORED
STM32F103RC_bigtree_512K_NOUSB  IGNORED
STM32F103RE_bigtree             IGNORED
STM32F103RE_bigtree_NOUSB       IGNORED
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
============================= 1 failed, 0 succeeded in 00:00:03.944 =============================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

steverichab, 2019-12-03:
I struggle a long time with the same issue until I found these instruction, they work fine:
https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/

patzik, 2019-12-03:
The problem seems to be cloning the GitHub.

salami738, 2019-12-08:
did you install git first?

huzargarden, 2020-02-02:
Same problem