bigtreetech/BIGTREETECH-SKR-mini-E3 #138: Doubts

MDFPereira, 2019-12-03:
Hello everyone.

Can someone from BTT explain this, please?
And also what changes you did to the firmware, since the new versions on GITHUB are dated from 8 days ago...

#
# BigTree SKR Mini V1.1 / SKR mini E3 / SKR E3 DIP (STM32F103RCT6 ARM Cortex-M3)
#
#   STM32F103RC_bigtree .............. RCT6 with 256K
#   STM32F103RC_bigtree_NOUSB ........ RCT6 with 256K (no USB)
#   STM32F103RC_bigtree_512K.......... RCT6 with 512K
#   STM32F103RC_bigtree_512K_NOUSB ... RCT6 with 512K (no USB)
#   STM32F103RE_bigtree .............. RET6
#   STM32F103RE_bigtree_NOUSB ........ RET6 (no USB)
#

Thank you very much.

demitrix, 2019-12-03:
Some boards shipped with the RET6 which is a 512k chip, most shipped with the RCT6 which is 256k but many people have had luck with flashing 512k to the chip anyhow. So the option is there to remove the 256k limit and build larger firmware if you wanted to try flashing it.

MDFPereira, 2019-12-03:
Thank you Demitrix.
Sorry I was not clear enough. I was asking about the USB “tag” on the names.
I have the RCT6 and managed to upload the “complete” firmware by removing the limit.
The USB question could be explained by BTT.
BTW, for me everything it is working correctly so far... Will see tomorrow when I assemble the board on the machine ;)
Thank you, again.

demitrix, 2019-12-03:
I may be incorrect but I believe NOUSB just removed the composite usb device. Enabling/disabled sd access over USB.

buhralex, 2019-12-03:
> I may be incorrect but ...

You're not :)

totti001, 2019-12-15:
Hi,

Does anybody tried to flash a firmware to a RCT6 board with the STM32F103RC_bigtree_512K option? My finding is that the bootloader checks the file size and does not flash it if it's more than 225k.