bigtreetech/BIGTREETECH-SKR-mini-E3 #139: BLTouch issue

MDFPereira, 2019-12-03:
Hello everyone.

I have a SKR Mini E3 V1.2.
I'm trying to compile the new firmware with bltouch.
The instruction from bltouch page says to uncomment
#define ENDSTOP_INTERRUPTS_FEATURE
When compiling I get this error:
#error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."

Two questions:
1- I'm not even using 2208 sonce the board comes with 2209 "on-board", where is this error coming from?
2- Will the BLTouch work without this feature?

I guess with a 32bit processor this will not make any difference because there is enough CPU. Am I wrong?

Thank you in advance.

bojanpotocnik, 2019-12-03:
You can compare your configuration with [my BLTouch branch](https://github.com/bojanpotocnik/Marlin/tree/bugfix-2.0.x_SKR-mini-E3-V1.2_BLTouch), for me it's working perfectly with `ENDSTOP_INTERRUPTS_FEATURE` and `ENDSTOPS_ALWAYS_ON_DEFAULT` enabled.

buhralex, 2019-12-03:
> #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."

I had the same problem. This is because in the current Marlin 2.0 Bugfix build the SanityCheck.h was changed. You need to change it to the following:

`#if HAS_TMC220x && !defined(TARGET_LPC1768) && !defined(TARGET_STM32F1) && ENABLED(ENDSTOP_INTERRUPTS_FEATURE) && !( \
   defined(X_HARDWARE_SERIAL ) || defined(X2_HARDWARE_SERIAL) \
|| defined(Y_HARDWARE_SERIAL ) || defined(Y2_HARDWARE_SERIAL) \
|| defined(Z_HARDWARE_SERIAL ) || defined(Z2_HARDWARE_SERIAL) \
|| defined(Z3_HARDWARE_SERIAL) || defined(E0_HARDWARE_SERIAL) \
|| defined(E1_HARDWARE_SERIAL) || defined(E2_HARDWARE_SERIAL) \
|| defined(E3_HARDWARE_SERIAL) || defined(E4_HARDWARE_SERIAL) \
|| defined(E5_HARDWARE_SERIAL) )
#error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
#endif`

sismo99, 2019-12-11:
> > #error "Select hardware UART for TMC2208 to use both TMC2208 and ENDSTOP_INTERRUPTS_FEATURE."
> 
> I had the same problem. This is because in the current Marlin 2.0 Bugfix build the SanityCheck.h was changed. You need to change it to the following:
> 
> Just open the SanityCheck.h and change **Line 2067** from this:
> `#if HAS_TMC220x && !defined(TARGET_LPC1768) && ENABLED(ENDSTOP_INTERRUPTS_FEATURE) && !( \`
> 
> to this:
> `#if HAS_TMC220x && !defined(TARGET_LPC1768) && !defined(TARGET_STM32F1) && ENABLED(ENDSTOP_INTERRUPTS_FEATURE) && !( \`

Just to share that it worked i had the sae proble and with this change worked perfectly

HVJ17, 2020-01-15:
It's not really needed on most setups.
Just make sure these two lines are both on or off together.

//#define ENDSTOPS_ALWAYS_ON_DEFAULT - in adv.h
//#define ENDSTOP_INTERRUPTS_FEATURE