bigtreetech/BIGTREETECH-SKR-mini-E3 #140: Unable to connect "See" the BTT SKR E3 Mini V1.2 to my Windows 10 USB.

steverichab, 2019-12-03:
I have just succeeded in compiling the new firmware in VSCode/PIO for my BTT E3 Mini V1.2 thanks to Morningreis on reddit.

https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/

 Unfortunately I am unable to flash the firmware to the board/sd card and I cannot even see the board on my PC.

I see the MapleDrv-Usb files on the big tree tech github repo and that could possibly be part of the solution but I don't really know how to install/use that.
I have tried simply moving posted the firmware.bin file to the sd card but that won't do anything either.

Any Advice would be appreciated

buhralex, 2019-12-03:
> I have tried simply moving posted the firmware.bin file to the sd card but that won't do anything either.

Have you followed exactly these steps?
- Copy the __firmware.bin__ file to the root directory of the sd card.
- Insert the sd card into the printer.
- In case your printer is currently powered on, shut it down by removing the power cord.
- Power on your printer. The screen will be blank until the flashing is done. This should not take more than 15 seconds. If your screen stays blank the firmware file is broken.

steverichab, 2019-12-03:
Thanks buhralex,
I have done all the steps you listed above but there is no "Flashing" it is as if the board doesn't see the card and neither does the PC see the board.
I have tried a second BTT SKR E3 Mini V1.2 and multiple SD cards all newly formatted and assorted sizes. They all behave the same => red light on, no flashing only a blank screen on the printer.
Again when the PC is connected to the E3 Mini, the PC does not even recognize that there is anything on the USB port.

brew99, 2019-12-03:
What size is the firmware.bin file on your SD card?

What default env. are you using in platformio.ini?

I had a V1.0 board that had a locked down bootloader, which wouldn't allow me to install anything above a filesize of 228kB.   It would skip the new firmware load and go straight to loading Marlin.

steverichab, 2019-12-04:
The default env. in platformIO.ini is STM32F103RC_bigtree and the bin file size that compiles in platformio is 212kB

I don't understand what you mean by going straight to loading Marlin I have tried both using the USB and moving the 212kB bin file to the sd card and booting neither works.

brew99, 2019-12-06:
I don't see anything wrong with that envir. and file size.   I would suggest removing all connections to the board except power and seeing if it responds.  Don't even connect the LCD for now.   You should be able to get that LED flashing red when it is loading the new firmware.   If it won't do that, then possibly a bootloader issue.

Have you been able to get the PC to identify the USB port at all?

steverichab, 2019-12-06:
Thanks brew, Yes that is correct I have never been able to connect my PC via USB to any 32 bit board( I have 2ea BTT SKR E3 Mini V1.2) and 1ea BTT SKR V1.3 and I would think I would be able to see one of these boards as a usb drive but nothing. I then tried to copy the firmware.bit to an SD Card still no card will boot from that card.

Given all this I thought it might be a driver on my PC.  On the BTT github there is a Maple driver but I am not sure what thats about.  

Getting a little desperate, I thought this was going to be a cheep and simple Christmas present upgrade but I can't get the minis to flash.

It appears I have got the VSCode/PIO learning curve behind me but still not able to flash the code on the board(s) .  Thanks for reaching out.

brew99, 2019-12-06:
Make sure that your SD cards are formatted FAT or FAT32.   Are they recognized when you use a USB to SD card adapter and plug them into your PC directly (i.e. not using a USB cable)?   The firmware loaded on the machine should not affect you being able to read the SD card.   weird that you have tried a few boards and none of them can load new firmware.bin from the SD

steverichab, 2019-12-07:
The sd cards are formatted as FAT 32. I see how the usb driver and the sd card are two separate issues.
If I but the usb drive aside and focus on the sd card just to be clear the only thing necessary is a firmware.bin file that is appropriate for the board in the root directory of the sd card. Is that right?
Is is possible to brick these BTT boards?

brew99, 2019-12-07:
Yes that’s right.  Only need the firmware.bin in root.  At this point I would totally remove everything and only have power connected.  Verify you are getting 24v from your power supply.   Turn on the power and if it loads the firmware one of the led will fast blink for about 10s.  

I don’t think you could brick the board by just loading firmware as it it writing to addresses above the boot loader.  If you want to dig in further, then you could get an stlink v2 device and read/write to the mcu

Curmudgeon101, 2019-12-11:
Same issue here. I received my E3 Mini V1.2 a few days ago. I also noticed on the board that F2 was hot. D19 became very hot when I attached to usb and caused usb power surge errors in Windows 7.

brew99's advice above worked for me. 

- Remove all board connectors except power.
- Turn on machine with firmware.bin loaded onto sd and watch for blinking red led

I reconnected the LCD and powered the printer to confirm it booted into marlin before adding any other connections. At this point Windows device manage would display the board as an unknown device under ports. Updating to the provided maple driver solved my com issues. 

It's also worth noting that I had a bltouch wired incorrectly to the servo connector on my very first power up.

steverichab, 2019-12-12:
Thanks for advice,  I have disconnect all connections to the BBT SKR Mini V1.2 except the power. I put in the SD Card with the firmware.bin file on it(both the 128Mb that came with the board and assorted other 2Gb & 8Gb).  
       When I connect it to a Windows 10 PC the red power light comes on but the PC does not "see " anything connected to the USB port.  I did this to a second BTT SKR Mini V1.2 and the same thing happens.
       When I connect a BTT SKR V1.3 to the same PC it "sees" the SD card and the files it contains. Note, there is a jumper on the SKR V1.3 that says get the power from the USB that must be in the correct position to make this work but there is not anything of the sort on the SKR Mini V1.2.

    I have come to the conclusion that as soon as I can "see" the Mini's  SD card from my PC I will be able to get this thing to work. Any ideas?

brew99, 2019-12-12:
@steverichab , in the configuration.h file, try and comment out

#define SD_CHECK_AND_RETRY

This might help you in "seeing" the SD card as mass storage.

Also, if you still can't get the firmware to load.   Disconnect everything from the board except power.   Insert the SD card with the firmware.bin (do not connect USB cable to board at this time) then power on.   If the board is working, you should see the red LED rapid flashing for about 10s.

Rhteam67, 2019-12-12:
Hello,

Same problem here, I managed to compile a firmware with platformio since marlin bugfix 2.0.
I put firmware.bin on my SD card and she installed marlin.

Now I just installed a bltouch so I take the firmware available by Bigtreetech and I put it on my pristine SD but the card does not want to install my firmware, it starts with the old firmware.

I have the impression that the boot loader does not want to read the SD

TerryFrench, 2019-12-12:
Try this:
- put the microSD card into your PC directly. Run checkdisk (right click on
drive in explorer, select properties, then tools, then "check/repair file
system").
Then delete the old .bin and .cur files.
You can now put some new firmware on the card and flash the printer.
do NOT have the usb cable plugged in when you flash the printer. Go from
power off for 30 seconds, put the card in, turn the power on, wait
30seconds to 1 minute for marlin to be flashed.

Note: for the weird power stuff, you need to block one pin on the USB port
(do that on the computer side) to not have the 5V provided by the PC to the
printer board. You can find online how to do that (google it with Ender 3)

There is currently a bug when after EVERY flash, the copy of the old file
(a Move really on FAT) fails in the middle leaving the FAT corrupted and
seeing 2 files, the .bin adn the .cur. (after flash .bin is renamed to .cur
normally).

Curmudgeon101, 2019-12-13:
Just to limit variables I'd suggest using the known working firmware.bin file in the bigtree repo https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/tree/master/firmware/V1.2 before testing one compiled yourself.

I think some have reported problems with shorting through a mounting screw or boss so leaving the board hanging while flashing would rule this out as well (just make sure it doesn't touch anything metal).  

I think the firware will need to be flashed correctly before any chance of seeing the SD as a mass storage device. As a curiosity what's the need to see the SD in Windows while it's in the printer?

@Rhteam67 just to cover bases, did you make sure to rename the bltouch file to "firmware.bin" on the SD card? A simple mistake that I made the first time I tried to load it.

steverichab, 2019-12-22:
Just wanted to let everyone know what the problem turned out to be.

Summery: 
Board #1 E3 Mini V1.2 purchased 10/09/19  Would not connect to Win 10 PC or load firmware.bin
Board #2 E3 Mini V1.2 purchased 11/24/19  Would not connect to Win 10 PC or load firmware.bin
Board #3 E3 Mini V1.2 purchased 12/16/19  Connects to Win 10 PC and can load firmware.bin from SD

All three boards came for "kingprint" and fortunately I was able to return the first 2 but only after hours of my and your time. I assumed it had to be something I didn't understand when Board #2 did the same as # 1.  
Thanks everyone, I really have learned a lot and look forward to finishing this printer.  I sill must use the SD card to update the firmware but at least I can update the firmware on Board #3.

VSCode still cannot upload but the firmware compiles well.   The PC "sees" the com port 13 is connected but I cannot see the files on the card over the USB cable.
Any ideas?
Thanks Again

brew99, 2019-12-22:
Ensure that SD_CHECK_And_RETRY (or something like that) is commented out

Rhteam67, 2019-12-26:
Bonjour,
Tout fonctionne pour flash, mais je quand le plateau bouge en Y vers l'avant, il tape contre le tendeur de courroie. Une idée ?
Ender 3 avec Tête E3DV6 créality

closed 2019-12-26 by steverichab

Interplain, 2020-01-05:
I have the same issue, the Board a SKR v1.3 was working lovely and crashed while using Simplify3d via usb print the Board disconnected and reconnected within a few seconds and then disconnected to be missing from windows.. Restarted the board and now Win10 will not see board neither card or com port, no notification with windows sound. Tried another USB port, nothing.

I have formatted the card with a USB Card reader Fat32 512kb and put back on the CUR file  that it was running on. I had the Red Jumper on the power from usb an it works still but no USB connectivity.  In marlin I have it -1 not zero an Baudrate at 115000. I have sent off for a new board to see if new board will be seen. If not it maybe a windows issue.

To be honest I am new to this board, previously came from Ramps A8 board.