bigtreetech/BIGTREETECH-SKR-mini-E3 #142: Unable to complie

Cactus127, 2019-12-04:
I can't even get it to build even without making any edits...

Terminal printout below

> Executing task in folder Marlin-2.0.x-SKR-Mini-E3-V1.2: C:\Users\Ben\.platformio\penv\Scripts\platformio.exe run <

Processing STM32F103RC_bigtree_NOUSB (platform: ststm32; framework: arduino; board: genericSTM32F103RC)
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------

Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/ststm32/genericSTM32F103RC.html
PLATFORM: ST STM32 5.7.0 > STM32F103RC (48k RAM. 256k Flash)
HARDWARE: STM32F103RCT6 72MHz, 48KB RAM, 256KB Flash
DEBUG: Current (blackmagic) External (blackmagic, jlink, stlink)
PACKAGES: toolchain-gccarmnoneeabi 1.70201.0 (7.2.1), framework-arduinoststm32-maple 1.10000.190819 (1.0.0), tool-stm32duino 1.0.2
Converting Marlin.ino
LDF: Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
LibraryManager: Installing U8glib-HAL
Using cache: C:\Users\Ben\.platformio\.cache\a7\b7a9b473d1c239d9843ca943887cd0a7
Unpacking...
FileNotFoundError: [Errno 2] No such file or directory: 'c:\\Users\\Ben\\Downloads\\BIGTREETECH-SKR-mini-E3-master\\BIGTREETECH-SKR-mini-E3-master\\firmware\\V1.2\\Marlin-2.0.x-SKR-Mini-E3-V1.2\\.pio\\libdeps\\STM32F103RC_bigtree_NOUSB\\_tmp_installing-vnv0vzjd-package\\U8glib-HAL-bugfix\\src\\clib\\HAL_example_files\\HAL_LCD_defines.h':
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 151:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Ben\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 605:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Ben\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 286:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Ben\.platformio\platforms\ststm32\builder\main.py", line 91:
    target_elf = env.BuildProgram()
  File "C:\Users\Ben\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 140:
    _build_project_deps(env)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\builder\tools\platformio.py", line 48:
    project_lib_builder = env.ConfigureProjectLibBuilder()
  File "C:\Users\Ben\.platformio\packages\tool-scons\script\..\engine\SCons\Environment.py", line 224:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 1027:
    project.install_dependencies()
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\builder\tools\piolib.py", line 859:
    lm.install(uri)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\managers\lib.py", line 301:
    force=force,
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 696:
    pkg_dir = self._install_from_url(name, url, requirements, track=True)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 480:
    self.unpack(dlpath, tmp_dir)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\managers\package.py", line 224:
    return fu.unpack(dest_dir, with_progress=False)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 145:
    self._unpacker.extract_item(item, dest_dir)
  File "C:\Users\Ben\.platformio\penv\lib\site-packages\platformio\unpacker.py", line 39:
    self._afo.extract(item, dest_dir)
  File "C:\Users\Ben\.platformio\python37\lib\zipfile.py", line 1602:
    return self._extract_member(member, path, pwd)
  File "C:\Users\Ben\.platformio\python37\lib\zipfile.py", line 1673:
    open(targetpath, "wb") as target:
=============================================================================================== [FAILED] Took 0.86 seconds =============================================================================================== 

Environment                     Status    Duration
------------------------------  --------  ------------
megaatmega2560                  IGNORED
megaatmega1280                  IGNORED
at90usb1286_cdc                 IGNORED
at90usb1286_dfu                 IGNORED
DUE                             IGNORED
DUE_USB                         IGNORED
DUE_debug                       IGNORED
LPC1768                         IGNORED
LPC1769                         IGNORED
sanguino_atmega644p             IGNORED
sanguino_atmega1284p            IGNORED
melzi                           IGNORED
melzi_optiboot                  IGNORED
rambo                           IGNORED
STM32F103RE                     IGNORED
STM32F103RC_fysetc              IGNORED
STM32F103RC_bigtree             IGNORED
STM32F103RC_bigtree_NOUSB       FAILED    00:00:00.862
STM32F103RC_bigtree_512K        IGNORED
STM32F103RC_bigtree_512K_NOUSB  IGNORED
STM32F103RE_bigtree             IGNORED
STM32F103RE_bigtree_NOUSB       IGNORED
STM32F4                         IGNORED
STM32F7                         IGNORED
ARMED                           IGNORED
STM32F103VE_longer              IGNORED
mks_robin                       IGNORED
mks_robin_lite                  IGNORED
mks_robin_mini                  IGNORED
mks_robin_nano                  IGNORED
jgaurora_a5s_a1                 IGNORED
STM32F407VE_black               IGNORED
BIGTREE_SKR_PRO                 IGNORED
BIGTREE_BTT002                  IGNORED
teensy31                        IGNORED
STM32F103CB_malyan              IGNORED
chitu_f103                      IGNORED
teensy35                        IGNORED
esp32                           IGNORED
fysetc_f6_13                    IGNORED
linux_native                    IGNORED
SAMD51_grandcentral_m4          IGNORED
include_tree                    IGNORED
========================================================================================== 1 failed, 0 succeeded in 00:00:00.862 ==========================================================================================
The terminal process terminated with exit code: 1

Terminal will be reused by tasks, press any key to close it.

brew99, 2019-12-04:
Are you connected to the internet?   Also, try closing off VScode then open again.   Is this the first time you are trying to compile the code within VScode?

steverichab, 2019-12-04:
Try looking at this information I got it to compile on my BTT SKR E3 Mini V1.2
https://www.reddit.com/r/ender3/comments/dojh3v/guide_for_those_upgrading_to_an_skr_e3_mini_v12/

TheGadge, 2019-12-05:
Keeping the whole folder for marlin etc at the root of c:\ and right clicking on the folder and removing all read only solved a LOT of problems for me and others I've suggested it to. If it still won't work, you need to downgrade platformio and it's core to previous version. (extensions>click platformio cog>install another version)

thisiskeithb, 2019-12-05:
> If it still won't work, you need to downgrade platformio and it's core to previous version.

This has been patched in PIO as well as Marlin, so downgrading is no longer required. Make sure you’re using the later supported IDE like VSCode/VSCodium and not Atom since PIO dropped support for it.

DoctorDanke, 2019-12-07:
I also had this problem, found the solution, and also helped a couple of other people.  Instead of rewriting the whole solution, here is a pasting of a previous post of mine:
-------------------------------
I also just experienced this problem, using VSCode and PlatformIO. After hours of not knowing what I was doing wrong, with a little research, I've found the fix, and the fix is irritatingly simple. It appears that some part of the Windows environment is encountering a filename or path name limit, and all you have to do is rename the folder/folders to something shorter than "BIGTREETECH-SKR-mini-E3-master". I went to the extreme and actually pulled the "BIGTREETECH-SKR-mini-E3-master\firmware\V1.2\Marlin-2.0.x-SKR-Mini-E3-V1.2" folder out of the main folder structure (to further reduce the path name size), renamed the "Marlin-2.0.x-SKR-Mini-E3-V1.2" folder to just "Marlin-2.0.x", and opened the renamed "Marlin-2.0.x" folder as a project in VSCode, and it now builds and works just as expected, and doesn't fail to build anymore.

It appears in the future, BTT is going to have to reduce the large number of characters in the folder names, to accommodate the Windows environment limits.

EliMcG, 2020-01-05:
DR Danke

 I have been searching all afternoon... THANK YOU !!!!

dan-youd, 2020-03-05:
Thanks DoctorDanke,

I've finally got around to installing this board today and needed to recompile firmware so that it knew I had a BLTouch and I got the exact same error. I renamed the folders and it worked first time.

I think BTT need to take a look at some of these smaller avoidable issues.

Rexxcalibur, 2020-03-07:
Dr Danke,

You are a lifesaver (or at least sanity saver)
why oh why did they do that?

FJJWFP, 2020-04-18:
Are you kidding me. How on earth did you possibly think of this. Path too long. Well at least we know how to compile them now. Funny thing, I could compile the 256k version, but not the 512k version, supposedly my path was "512k" (4 characters) too long.

Gearhead-42, 2020-04-25:
Thanks Doc, brilliant find.  Had me pulling my hair out til I came across this!