bigtreetech/BIGTREETECH-SKR-mini-E3 #143: I need help with bltouch clone

Smacken90, 2019-12-04:
Hi I have an ender 3. skr mini 1.2. I have a bltouch clone. I can't make it work properly. I would be grateful for any help

I have made the following changes

Uncommented #define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN.
Uncommented #define BLTOUCH.
EDIT: Uncommented #define ENDSTOP_INTERRUPTS_FEATURE.
Set the appropriate values to X_PROBE_OFFSET_FROM_EXTRUDER. Y_PROBE_OFFSET_FROM_EXTRUDER and Z_PROBE_OFFSET_FROM_EXTRUDER.
Uncommented #define AUTO_BED_LEVELING_BILINEAR (or use your preferred ABL method).
Uncommented #define Z_SAFE_HOMING. Although this is not required, it avoids damaging your printer by ensuring that Z homing is always done inside the print bed area.

bojanpotocnik, 2019-12-05:
1) How did you connect it to the board?
2) How does it (not) work?

Smacken90, 2019-12-05:
I have it in servo. 
It is very irregular. It does not go in and out as it should. It often misses.
Uncommented #define BLTOUCH_SET_5V_MODE 

I've tried  Uncommented  BLTOUCH_FORCE_SW_MODE
and playd aorund with  BLTOUCH_DELAY 500
nothing seems to help
Thank you so much for helping me :)

smcallis77, 2019-12-06:
@Smacken90 What version of the BLTouch do you have? What version of the SKR Mini do you have? I have BLTOUCH V3.1 and SKR MINI E3 v1.2.

Make sure the wiring to GND, 5V and PA1 are correct. I had to swap a couple of the pins on my bltouch (version 3.1) to get the pins in the right order.

If you define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN then the probe sense cable needs to go into Z_STOP socket. (For me, I actually redefined the Z_STOP pin to PC14 instead of PC2 in the HAL definition file - pins_BTT_SKR_MINI_E3.h, disconnected the Z_STOP switch and plugged the BLTouch probe pin into the PROBE socket instead - for me it was just easier to make the change in firmware rather than pull the printer apart).

The only config change I made that is not the same as you is I haven't defined ENDSTOP_INTERRUPTS_FEATURE. I missed that. It should probably be defined. I will fix that later. My BLTouch works 100%, I have never had a failed probe.

I have heard the v1.0 boards have some problem with a noisy 5V rail. Not sure if it affects SERVOS, Z_STOP, PROBE or any combination of the three. I heard that some people had had success with a capacitor at the relevant point to filter out the noise. The issue seems to have been fixed on the V1.2 boards. Hope this info helps.

tatusah, 2019-12-07:
BTT previously recommended enabling #define FAN_SOFT_PWM when using BLTouch. Have you tried this?

closed 2019-12-07 by Smacken90

Smacken90, 2019-12-07:
I have a skr mini e3 1.2 The prob is a Fystec. https://www.aliexpress.com/item/32838356613.html.
 smcallis77 .Make sure the wiring to GND, 5V and PA1 are correct. I had to swap a couple of the pins on my bltouch (version 3.1) to get the pins in the right order. Checked and clear :)

If you define Z_MIN_PROBE_USES_Z_MIN_ENDSTOP_PIN then the probe sense cable needs to go into Z_STOP socket. (For me, I actually redefined the Z_STOP pin to PC14 instead of PC2 in the HAL definition file - pins_BTT_SKR_MINI_E3.h, disconnected the Z_STOP switch and plugged the BLTouch probe pin into the PROBE socket instead - for me it was just easier to make the change in firmware rather than pull the printer apart). I have tried both pc14 and pc2. I get the same result.

tatusah 
BTT previously recommended enabling #define FAN_SOFT_PWM when using BLTouch. Have you tried this?   I'll try that feature

Thanks so much for your quick answers! 👍