bigtreetech/BIGTREETECH-SKR-mini-E3 #144: BigTreeTech Smart Filament Detection Module not triggering on when connected to PT-DET

TerawattX, 2019-12-05:
I have configured by board and connected the Smart Filament Detection Module per the provided instructions in the Github repository, however as soon as the first layer has started it will indicate there is no filament in the machine and launch the M600 command.  

This issue was also reported by another user on the Filament Detection Module repo:(https://github.com/bigtreetech/smart-filament-detection-module/issues/1)

They were able to confirm signaling from the module, but the board was not sensing it.

pa0los, 2019-12-06:
PT-DET is meant for switching the power supply.
For filament runout you should use E0-STOP connector (see the pin defines file in Marlin for SKR mini E3).

TerawattX, 2019-12-06:
That certainly explains the issue, though I could have sworn I read you could change the pins in the config to use PC12 vs PC15.
 
Question: The detection module ships with a 3 to 4 pin cable, E0-Stop is a 2 pin connector.  Will the Filament Detection module still operate properly on 2 pins vs 3?  I know this is a different product from the SKR board, but since both are BigTreeTech products I figured it wouldn't hurt to ask.

TerawattX, 2019-12-20:
So I'm going to reopen this because the documentation isn't very clear on the issue.  The [PDF](https://github.com/bigtreetech/smart-filament-detection-module/blob/master/manual/smart%20filament%20sensor%20module%20manual191205.pdf) for the module simply states to use "any extended interface."  I took this to indicate that PT-DET would work as long as you set the interface accordingly in the pins_BTT_SKR_MINI_E3.h file.

There is also a comment by another user that they were able to get theirs working on PT-DET by setting FIL_RUNOUT_INVERTING to true, however when I do this it simply is always open and never triggers when it runs out of filament.

Is there a clearly defined list of which interfaces are valid?  As I stated earlier, I don't think E0-Stop would work because the module requires a SIN, VDD, and GND wire, and I'd use the Servos interface, but that's where my BLTouch is hooked up.  Any alternatives or confirmation one way or another about PT-DET?

btsubjectdata, 2020-01-23:
Mine doesn't work either. Same exact issue. Please advise.

usk-johnny-s, 2020-02-25:
I found root of this issue.

At following line will change a PortC-12 to Output-LOW. (C-12 is connected PT-DET.)
https://github.com/bigtreetech/BIGTREETECH-SKR-mini-E3/blob/5123ce60d1f7a88e9e6686934db38046e9360b3f/firmware/V1.2/Marlin-2.0.x-SKR-Mini-E3-V1.2/Marlin/src/Marlin.cpp#L870

For resolv this issue, modify maple configuration.
Modify at L.102 -103 in "STM32F1/variants/generic_stm32f103r/board/board.h"
> #define BOARD_USB_DISC_DEV      NULL
> #define BOARD_USB_DISC_BIT      NULL

If modify maple is difficult, quick workaround is re-initialize PortC-12 after "MYSERIAL1.begin(BAUDRATE);".
Insert these lines at L.873 in Marlin.cpp
>  #if HAS_FILAMENT_SENSOR
>    runout.setup();
>  #endif

But this has PortC-12 Output LOW while very short time. it will cause level conflict If external device outputs high logic level.

In Detail...
"MYSERIAL1" is defined as USBSerial of maple framework.
This BIGTREETECH-SKR-mini-E3  is chose generic_stm32f103r as  maple framework configuration.

"BOARD_USB_DISC" is defined as PortC-12 at 
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/variants/generic_stm32f103r/board/board.h#L102

Followings are  indicate to set a PortC-12 to Output and LOW. https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/cores/maple/usb_serial.cpp#L75
and 
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/0383384f4cc91aadee1501e11bf08dd3e62d9fb9/STM32F1/cores/maple/libmaple/usb/stm32f1/usb_cdcacm.c#L387

qwewer0, 2020-05-19:
@usk-johnny-s  Can't find any that you mentioned. If this is still a thing, then could you point me to the right direction?
What should I download/change?

usk-johnny-s, 2020-05-20:
Unfortunately, Following issue close but nothing fixed.
https://github.com/rogerclarkmelbourne/Arduino_STM32/issues/759

In my build environment, It's on Windows 10 and based on SKR's how to.
First time build Marlin on VS code, platformio down loads ralated packages.
"STM32F1/variants/generic_stm32f103r/board/board.h" that is mentioned previous my post, is located "%USERPROFILE%\.platformio\packages\framework-arduinoststm32-maple\STM32F1\variants\generic_stm32f103r\board\board.h".
I directly edited it and then re-build Marlin on VS code.

qwewer0, 2020-05-20:
Thanks for the detailed explanation, got it now.
Do you think that it can have a side effect?

usk-johnny-s, 2020-05-21:
At L.102 -103 in "STM32F1/variants/generic_stm32f103r/board/board.h",
To modify to NULL is no side effect on SKR-mini-E3, I think.
Because, at SKR-mini-E3, pin "GPIOC-12" it previous definition, is just connected to PT-DET only, not assigned any function at normal.

At original maple, USB_DISC signal cause to drives USB-DP to low forcely, it cause force disconnect USB interface from host.
"maple" is "https://www.leaflabs.com/maple", and schematics is in "https://github.com/leaflabs/maple".

USB_DISC is described in Maple framework README.
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/9d46a1c27d4dc9dd4df06b76a1d81684519d8acc/STM32F1/cores/maple/libmaple/usb/README#L49-L51

* Initialization of USB_DISC.
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/b5cd37696b7057f8489c0a17801420756b44a4e7/STM32F1/libraries/USBComposite/usb_generic.c#L313-L316

* Assertion of USB_DISC.
https://github.com/rogerclarkmelbourne/Arduino_STM32/blob/b5cd37696b7057f8489c0a17801420756b44a4e7/STM32F1/libraries/USBComposite/usb_generic.c#L412-L414

Cheule, 2020-06-27:
TL;DR  PT-DET port is not recommended for 3 wire filament sensors.
----
Using a TH3D 3 wire filament sensor, I found that the PT-DET port was very flaky on my board. When the filament sensor would close (filament out) the LCD display (stock 12864) would blank/glitch/freeze.

I ordered another e3 Mini v1.2 board and it had the same problems. Finally I found out that using the PC12 pin on PT-DET is not wise on this board. Instead using PC15 on the E0-STOP port is recommended. I could have just wired the Black wire (GND) and white wire (S) from my sensor to the E0-STOP port, and ignored the red wire (5v) but I didn’t want to lose my LED confirmation of inserted filament. So I attached the red filament sensor wire to the 5v of PT-DET port.

[Picture of red/white/black wire placement on the BTT SKR e3 mini v1.2 board](https://imgur.com/gallery/VEO6Hs2)

In Marlin, don’t forget to #define FIL_RUNOUT_PIN PC15

usk-johnny-s, 2020-07-03:
PC12 is 5V tollerant, but some one is not.
Check out connecting sensor output level comply with MCU pin, if you wish to use other pin.

PC12:I/O Level is indicated "FT". ( Five Volts Tollerant)
https://www.st.com/resource/en/datasheet/stm32f103rc.pdf#page=35

PC15:I/O Level is nothing indicated. (NOT Five Volts Tollerant)
https://www.st.com/resource/en/datasheet/stm32f103rc.pdf#page=31

Cheule, 2020-07-03:
> PC12 is 5V tollerant, but some one is not.
> Check out connecting sensor output level comply with MCU pin, if you wish to use other pin.
>
> PC12:I/O Level is indicated "FT". ( Five Volts Tollerant) [...]

This explains a lot. And is very helpful. 5v was definitely what was being sent to PC12 when the LCD display corrupted.

usk-johnny-s, 2020-07-06:
Watch out difference between I/O Level "FT"  or not, about "Vin" of  "Absolute maximum ratings".
https://www.st.com/resource/en/datasheet/stm32f103rc.pdf#page=43