bigtreetech/BIGTREETECH-SKR-mini-E3 #145: [BUG] No customer service

hedwards, 2019-12-07:
Tech support doesn't seem to actually address problems beyond a few standard questions to appear like they're supporting the product. I deeply, deeply regret wasting my money on this product as it was broken from first install and tech support has been more concerned with figuring out how to avoid replacing or refunding the product than fixing the problem.

Even wiring the device and using the firmware that tech support provided has done nothing, but I'm still not getting a refund/replacement and they haven't provided any further assistance in weeks.

If they're going to sell these boards, they should disclose that there are no instructions included and that the firmware may or may not work and if it doesn't, they aren't going to provide any meaningful help in the process. 

I deeply regret wasting my money on this hunk of junk.

nicstoun, 2019-12-07:
I have   same experience , they replace me two boards, second burn my drivers ..now send third , i buy first  in August ..  still use stock noise board.  Please copy this on BIGTREETECH groups.

hedwards, 2019-12-07:
Anyways, Aliexpress was completely useless in terms of doing anything to help me. They deemed me not to have made my case that the board didn't work, despite having uploaded video of it powering on, not displaying anything on the LCD screen and the status light not coming on.

I've reported the fraud to my credit card issuer, hopefully, they'll claw the fraudulently obtained money back and under no circumstance will I be buying anything from Aliexpress in the future if they're willing to allow this scam to continue. I'm on the hook for the money, even though BTT couldn't get the board working and ultimately stopped responding to requests for support and refused to replace the broken board or refund my money.

Dracrius, 2020-01-11:
@hedwards if you spoke to support like you are here I wouldn't want to help you either its a $30 mainboard and you know what you never should have expected support for anything you bought off Aliexpress thats like expecting the dollar store to give you support on the crappy ear buds you bought! LMFAO

Next time order direct from Bigtreetech I have heard of at least a few better interactions with support! Just be happy you didn't buy from TH3D they have no excuse at the 3x price they charge. Their are countless stories of the exact same kinda of support you have described. Actually most 3D printer manufactures do that. It is because they are not big giants the 3d printing community is still small. In these cases you catch more flies with sugar then vinegar.

I'd suggest reaching out with a more apologetic and problem solving attitude and less of a its everyones fault but mine mind set. Like jesus you posted this crap in a bug report on a github. Do you really think a single person maintaining this build is going to have anything to do with your hardware??? Cause they don't, what about an open source software page makes you think this is the place to rant about how bad you are with support! If you haven't figured it out you consistently get worst customer service when your rude to the person who can help you and it is VERY normal to run through all the cases of user error before doing a return. Sure it suck to hear the classic "did you try turning it off and on again" but thats their job not to just except you saying somethings broken. These companies do not have the revenue to be that fast and loss with returns especially these boards that are already sold at bottom dollar.

hedwards, 2020-01-11:
Dracrius, you're not helping. I posted here after I sought support via both their Aliexpress and their official support pages. Because people should know that the documentation isn't correct and that they weren't providing the bare minimum documentation needed to get the board working. You're self-righteous attitude is not helping anybody. Clearly, BTT doesn't agree with you because they sacked the support staff that just ignored those emails when it became clear that the board wasn't even powering on.
I have no problem with them making such basic suggestions, what I do have a massive problem with is that it hit the point where they had literal video of the device not working and just stopped responding.

Dracrius, 2020-01-11:
So your telling me that support eventually did come through and it was just one bad enployee?? I now truely stand by my point and I'm happy i said something as you were really painting the entire situation wrongly. It is truely unfortunate that you fell victum to a bad employee but the company as a whole has made good on that mistake from the sounds of it. Also the tiny box with a duck and a buisness card not having a manual is no damn surprise. Their site has all said information and links to this github which has more details then any manual you'll ever get with a closed source product. There are step by step guides right here!

hedwards, 2020-02-02:
Even before I contacted support it was a bad experience due to the poor quality of documentation and the apparently unreliable firmware that I couldn't compile or load. It should have been very quick for me to know whether or not firmware was loading and being asked to provide video documentation without clearly indicating what was needed was rather ridiculous. I get that English isn't their first language, but if that's going to be the procedure that decides what to do about it, there's no excuse for them not knowing the necessary English to communicate that. It wouldn't exactly be unexpected to need to express that.

I may buy from them in the future, but the whole experience really soured me on the prospect.

closed 2020-02-02 by hedwards