bigtreetech/BIGTREETECH-SKR-mini-E3 #146: [BUG] Permanent Blue Screen

TeLiP, 2019-12-07:
Tried to flash Bigtreetech's firmware for the BLTouch to the SKR Mini e3 V1.2.
The screen stayed permanently blue, no text. Waited about ten minutes before resetting. 
No previous working firmware builds will flash now. Screen is permanently blue.

### Steps to reproduce

1. Placed Bigtreetech's firmware.bin file onto SD card.
2. Powered on the printer
3. Permanent blue screen

Scope666, 2019-12-10:
I'm still waiting for my E3 to arrive from China, but I've had this EXACT same symptom happen multiple times on the stock board.  It happens when you get close to 99% used program space, the calculations used to determine if it fits or not seem to be off and not account for the bootloader.  (firmware flash too big and "steps on" the bootloader)

When this happens I have to use my Pi's GPIO pins and reflash it that way, which bypasses the need for a bootloader.  I'm not sure if that would apply for the SKR since it loads firmware off SD instead...

Procedure I use:   https://www.fission3d.com/guides/flash-bootloader-and-install-firmware-with-raspberry-pi

buhralex, 2019-12-12:
Your firmware is broken
That's not a bug. That's pebkac

HVJ17, 2020-01-01:
The issue is the wires on the BLTOUCH. There are several wiring variants. Switch the red and blue/brown on the end that plugs into SERVOS,  leave YELLOW alone.

Scroll to the very bottom of this https://www.antclabs.com/bltouch page to see BLTOUCH example.